plcpeople/nodeS7 #2: Quickly tested out this module and received error making ISO-on-TCP Connection

dlbrown06, 2015-02-17:
I just ran this test against an local s7 we have. The only thing I changed from the example code was the IP. I'm not familiar with ISO-on-TCP. Am I missing something on the configuration?

```
[97849,313122985] Initiate Called - Connecting to PLC with address and parameters:
[97849,313985846] { port: 102, host: '10.2.202.145', rack: 0, slot: 1 }
[97849,315746540] Connection cleanup is happening
[97849,329727559 10.2.202.145 S1] Attempting to connect to host...
[97849,334747742 10.2.202.145 S1] TCP Connection Established to 10.2.202.145 on port 102
[97849,334815782 10.2.202.145 S1] Will attempt ISO-on-TCP connection
[97849,341580313 10.2.202.145 S1] ISO-on-TCP connection DISCONNECTED.
Error - TCP connected, ISO didn't
```

plcpeople, 2015-02-17:
What type or Siemens part number of S7 PLC are you connecting to?

My first guess from that output would be that it's a 300 type and you should use slot 2 instead of slot 1.

dlbrown06, 2015-02-17:
Closer

```
[919,883304001] Initiate Called - Connecting to PLC with address and parameters:
[919,884208285] { port: 102, host: '10.2.202.145', rack: 0, slot: 2 }
[919,886065128] Connection cleanup is happening
[919,900610458 10.2.202.145 S2] Attempting to connect to host...
[919,908272737 10.2.202.145 S2] TCP Connection Established to 10.2.202.145 on port 102
[919,908350279 10.2.202.145 S2] Will attempt ISO-on-TCP connection
[919,913547538 10.2.202.145 S2] ISO-on-TCP Connection Confirm Packet Received
[919,918113432 10.2.202.145 S2] Received PDU Response - Proceeding with PDU 240 and 2 max parallel connections.
undefined

Process finished with exit code 0
```

dlbrown06, 2015-02-17:
Looks like the problem I was having was here upon connection:
```
if (typeof(err) !== "undefined") {
        // We have an error.  Maybe the PLC is not reachable.  
        console.log(err);
        process.exit();
    }
```

initiateConnection returns a callback with no parameters defined, so it is failing out here when it actually was connected.

I worked around it by changing `if (typeof(err) !== "undefined") {` to `if (err) {` so `undefined` is still a `falsey`

plcpeople, 2015-02-17:
Is it working for you now?  Slot 2 seems to work better.

dlbrown06, 2015-02-17:
Yeah... big fail on my part. I noticed you even did the due diligence to comment it within the code... :/

But yes, its working now. Thank you for contributing this great node module to the community.

closed 2015-02-17 by dlbrown06