plcpeople/nodeS7 #3: Datatypes

sembaye, 2016-05-24:
Hi,
I want to extend the functionality to read/write the remaining basic Datatypes (S5TIME,TIME,CHAR,DATE,TOD). In which places of the src-code do i have to do changes?
Need some hints!!

Thanks

Lichtjaeger, 2016-05-31:
I would like this.
Look from line 1529 to the end of the file.

Please complete all typs and aliases.
I miss:

- BOOL (implemented as X)
- LWORD
- SINT
- USINT (implemented as BYTE)
- UINT
- UDINT
- LINT
- ULINT
- LREAL
- S5TIME
- TIME (implemented as DINT)
- LTIME
- DATE
- TOD
- LTOD
- DT
- LDT
- DTL
- WCHAR
- WSTRING

gfcittolin, 2016-05-31:
@Lichtjaeger I agree with you about modularizing this package. I'm also thinking about making this package behave more like a library, by better controlling logging and proposing some API changes. I'll try to make a pull request on the next few days :)

plcpeople, 2016-06-01:
I wish I had the time to modularize this as this was my first NPM package and I have learned a lot since.  I will post some info on adding datatypes in the coming days (I am travelling this week) but I welcome a major pull request to clean some of this up.

Lichtjaeger, 2016-06-07:
In my opinion this is an important issue. I don't like that it's closed.

sembaye, 2016-06-07:
Oki doki

sagardubal, 2016-09-17:
We have been using the Code, everything is working well except for that we are not able to access I/O directly by using "Q0.0" or "i0.0". Kindly advice.

plcpeople, 2016-09-18:
Can you please post a log of the error that you get?  And what model of PLC are you using?  Q0.0 and I0.0 should "just work",

plcpeople, 2016-10-22:
I tested inputs and outputs (both reading and writing, writing only to inputs that were not defined in hardware configuration) and it worked fine on a 315-2 PN/DP.  Version 0.1.10 fixed some boolean write issues - that may have been an issue.  Please test again.

klaus0401, 2018-11-27:
lreal can just use readDoubleBE while for Lint can ether modify buffer function in Nodejs  or just make a new function in nodes7.