plcpeople/nodeS7 #4: Error on writing more then 32 byte of data

Lichtjaeger, 2016-06-01:
Hello,

I've got the following log:

> [18193,176543379] Initiate Called - Connecting to PLC with address and parameters:
> [18193,182467207] { host: '192.168.36.94', port: 102, slot: 1, rack: 0 }
> [18193,209406185] Connection cleanup is happening
> [18193,216527877 192.168.36.94 S1] Attempting to connect to host...
> [18193,224857695 192.168.36.94 S1] TCP Connection Established to 192.168.36.94 on port 102
> [18193,227583800 192.168.36.94 S1] Will attempt ISO-on-TCP connection
> [18193,235965423 192.168.36.94 S1] ISO-on-TCP Connection Confirm Packet Received
> [18193,242886738 192.168.36.94 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
> [18193,247310183] Translation OK
> [18193,249174666 192.168.36.94 S1] Preparing to WRITE g0,x0,v0,g1,x1,v1,g2,x2,v2,g3,x3,v3,g4,x4,v4,g5,x5,v5,g6,x6,v6,g7,x7,v7,g8,x8,v8,g9,x9,v9,g10,x10,v10,g11,x11,v11,g12,x12,v12 to value 50,3000,33,21,10000,100,10,30000,100,5,1000,0,20,-22000,75,10,30000,75,20,-22000,75,10,30000,75,20,-22000,75,10,30000,75,10,10000,25,11,0,10,0,0,0
> [18193,286220953] PDU type (byte 8) was returned as 2 where the response PDU of 3 was expected.
> [18193,288932884] Maybe you are requesting more than 240 bytes of data in a packet?
> [18193,292037751] <Buffer 03 00 00 13 02 f0 80 32 02 00 00 00 02 00 00 00 00 85 00>
> [18193,294749193] ConnectionReset is happening
> [18193,297145409] We Caught a read/write error ECONNRESET - will DISCONNECT and attempt to reconnect.
> [18193,300186252] ConnectionReset is happening
> [18194,791888445 192.168.36.94 S1] WRITE TIMEOUT on sequence number 2
> [18194,798587879] We must have timed Out - we have no response to process
> [18194,804174488] Stopping Processing Write Response Packet due to unrecoverable packet error
> [18194,814658604 192.168.36.94 S1] WRITE TIMEOUT on sequence number 3
> [18194,819903105] We must have timed Out - we have no response to process
> [18194,823505981] Stopping Processing Write Response Packet due to unrecoverable packet error
> true
> [18194,831520084] ResetNOW is happening
> [18194,833264341] Clearing an earlier scheduled reset

I want to write to an Array of Struct in DB2. One Struct (line) conatins g: USINT; x: DINT and v: INT. The addresses are map the following way:

```
    program.forEach((line, index) => {
        map['g' + index] = [`DB2,BYTE${index * 8}`, line.g]
        map['x' + index] = [`DB2,DINT${index * 8 + 2}`, line.x]
        map['v' + index] = [`DB2,INT${index * 8 + 6}`, line.v]
    })
```

I can write exactly four lines at a time. One byte more and I get the error.

There is something about 240 byte in the log but 4 * 8 byte = 32 byte is much lower then 240 byte.

I am using a S7 1214 v2.2 CPU.

plcpeople, 2016-06-01:
For some reason it looks like the code is not splitting up the packet.  240
bytes is the limit including all the protocol overhead so it's possible you
are exceeding it.  It is also possible that we are exceeding some limit of
a maximum number of separate writes in a packet that I did not know
existed.  Try setting, on line 55 of nodes7.js, the requestMaxPDU to 120
and see if the problem goes away.

If you send a Wireshark trace I can have a look at that also.  Wireshark
has an S7 communication plug-in for looking into problems like this.

If you don't want to do that or can't do that (packet sniffers like
Wireshark are illegal at some companies) that is fine, I can simulate this
on an S7-1500 next week but this week I am travelling and do not have
access to the PLC.

Regards,

Dana

Lichtjaeger, 2016-06-02:
Contains my node project, plc prints, cmd log and two wireshark files (for requestMaxPDU = 960 and 120).

[s7-node-test.zip](https://github.com/plcpeople/nodeS7/files/294940/s7-node-test.zip)

plcpeople, 2016-06-02:
It will likely take until the weekend for me to get to look at this. But I
will.

closed 2016-06-15 by plcpeople

Lichtjaeger, 2016-06-16:
Thanks. Woks.

JhalakJaviya, 2016-09-23:
Hi,

I've got the similar error

> [47627,634468465] Initiate Called - Connecting to PLC with address and parameters:
[47627,639772034] { port: 102, host: '192.168.0.123', rack: 0, slot: 1 }
[47627,680456488] Connection cleanup is happening
[47627,686857950 192.168.0.123 S1] Attempting to connect to host...
[47627,758235869 192.168.0.123 S1] TCP Connection Established to 192.168.0.123 on port 102
[47627,758744897 192.168.0.123 S1] Will attempt ISO-on-TCP connection
[47627,775050023 192.168.0.123 S1] ISO-on-TCP Connection Confirm Packet Received
[47627,824989767 192.168.0.123 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
[47627,825920631] Translation OK
[47627,827753308 192.168.0.123 S1] Preparing to WRITE TEST7 to value 666,777
[47627,847024306 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47627,867028889] PDU type (byte 8) was returned as 2 where the response PDU of 3 was expected.
[47627,867426845] Maybe you are requesting more than 240 bytes of data in a packet?
[47627,867684389] <Buffer 03 00 00 13 02 f0 80 32 02 00 00 00 02 00 00 00 00 81 04>
[47627,871380138] ConnectionReset is happening
[47627,951553257 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47627,952192893 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,53546918 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,53804738 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,153336671 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,153720143 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,254576684 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,254986352 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,355813829 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,356272913 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,457305302 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,457691115 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,557490107 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,558571512 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,659565117 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,659986713 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,761190666 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,761610414 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,862591479 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,863013699 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47628,964124413 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47628,964515661 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47629,65363334 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47629,65752050 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47629,166594503 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47629,166985379 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47629,267570732 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47629,267701136 192.168.0.123 S1] Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.
[47629,349585515 192.168.0.123 S1] WRITE TIMEOUT on sequence number 2
[47629,353276608] We must have timed Out - we have no response to process
[47629,353460280] Stopping Processing Write Response Packet due to unrecoverable packet error
SOMETHING WENT WRONG WRITING VALUES!!!!
Done writing.
[47629,367218249 192.168.0.123 S1] Unable to read when not connected. Return bad values.
[47629,369775618 192.168.0.123 S1] Adding TEST1,TEST4
[47629,371109538 192.168.0.123 S1] Adding TEST6
[47629,377080884 192.168.0.123 S1] Attempting optimization of item DB1,REAL8 with DB1,REAL0.20
[47629,383619050 192.168.0.123 S1] Not Sending Read Packet because we are not connected - ISO CS is 0
[47629,385735119] ResetNOW is happening
[47629,385906011] Clearing an earlier scheduled reset
[47629,386161731 192.168.0.123 S1] READ TIMEOUT on sequence number 3
[47629,388147492] Processing an undefined packet, likely due to timeout error
[47629,388357312] Timeout error - zero length packet
[47629,388435444 192.168.0.123 S1] Received a ZERO RESPONSE Processing Read Packet due to unrecoverable packet error
[47629,388501864] Processing an undefined packet, likely due to timeout error
[47629,388619284] Timeout error - zero length packet
[47629,388732396 192.168.0.123 S1] Received a ZERO RESPONSE Processing Read Packet due to unrecoverable packet error
SOMETHING WENT WRONG READING VALUES!!!!
{ TEST1: 'BAD 255',
  TEST4: 
   [ 'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255',
     'BAD 255' ],
  TEST6: 'BAD 255' }

commit 488447d7 (2016-06-15): Fixed #4