plcpeople/nodeS7 #5: Error on writing Array of Boolean

Lichtjaeger, 2016-06-03:
Hi again,

now I have a problem on writing an Array of Boolean. In reality I need 16bit but for testcase I used 2.

Adress is "DB1,X2.0.2". Write command: `plc.writeItems(['boolean'], [true, false], ...`

Log:
> [10622,237878058 192.168.36.94 S1] Preparing to WRITE boolean to value false,true
> [10622,292730964] Received write error of 7 on DB1,X2.0.2

[2bit_in_Array.pcapng.gz](https://github.com/plcpeople/nodeS7/files/297208/2bit_in_Array.pcapng.gz)

gfcittolin, 2016-06-03:
Hi @Lichtjaeger, from the dump you attached, we can see the PLC did not like your data type when writing:
[image]

Is the data type configured in the S7 an array of BOOL? Perhaps, try other data types, and let us know if it works :)

Lichtjaeger, 2016-06-03:
It is "Array[0..15] of Bool" with Offset 2.0.

If I adress a single Bool of this array I can write without any problem (thats my actual workaround).

I also tested arrays of other types (for example `'DB1,REAL16.2'`) and they are working.

plcpeople, 2016-06-03:
Will it let you write a length of 8 or 16?

Lichtjaeger, 2016-06-03:
Same Error.
Tried:
"DB1,X2.0.16" with length of 16
"DB1,X2.0.16" with length of 8
"DB1,X2.0.8" with length of 8

plcpeople, 2016-06-03:
Will it let you write an integer in DB1.INT2?  I think your PLC is just
doing data type checking that other PLCs don't do.  Is it an S7 1200?  Can
you write an array of boolean to an address defined as an integer?

Same Error.
Tried:
"DB1,X2.0.16" with length of 16
"DB1,X2.0.16" with length of 8
"DB1,X2.0.8" with length of 8

—
<https://github.com/plcpeople/nodeS7/issues/5#issuecomment-223550582>, or mute
the thread

Lichtjaeger, 2016-06-03:
It is the same PLC I used in #4 : S7 1214 v2.2

I CAN write a single boolean into an INT.
I CAN write an INT over the Array[0..15] of Bool.
I CAN NOT write an array of boolean to the PLC. Not on an INT and not on the Array[0..15] of Bool.

The 1200 should be similar to the 1500.

Lichtjaeger, 2016-06-08:
Is this working for you? Can you write an array of boolean to your PLC?

plcpeople, 2016-06-08:
I am travelling and have not had a chance to test.  This weekend I'll be
home.

plcpeople, 2016-06-15:
OK I am committing a fix, please test at your convenience and let me know.

closed 2016-06-15 by plcpeople

Lichtjaeger, 2016-06-16:
Works like expected. I like it.

But your npm-module now contains a test3.js, a test4.js and a test5.js file.

PS: Another question: It's not documented but `nodes7.removeItems()` (without arguments) removes all items. Or am I wrong?

plcpeople, 2016-06-16:
Good to hear it works.

Good catch - I will remove the extra files and update the docs and
re-publish later tonight.

commit 0f5ab262 (2016-06-15): Fixes #5