plcpeople/nodeS7 #12: net-module does't react when pulling of the cable (embedded Linux)

sembaye, 2016-08-10:
Hallo everyone,

i have a problem that I want to discuss.
The NodeS7-module uses the net-module to create a socket for the whole communication.

I'm reading cyclic values from the plc. To simulate a connection error I pulled off the Ethernet cable. I tested the reaction of my software on three OS. Windows, Ubuntu VM and the destination system an embedded Linux OS. Each of them behave different. The Windows-OS and Ubuntu VM are stopping the communication which is good. In the emb. Linux nothing happens. I debugged the net module to see what's happening in it.

The following picture shows the log-files of each OS.
[image]

left shell=Windows: after a Timeout of the nodeS7-library the net-module detects an ECONNRESET
middle shell=Ubuntu: after a Timeout of the nodeS7-library the net-module detects an EOF
right shell=embedded Linux: **nothing is getting detected**

When you have a look on the src-code of the net-module you can see that the debug-log onread is getting created within the onread-function. The on-read function is always called whenever the handle gets a buffer, or when there's an error reading.

Well, I have a reading error. What I am doing wrong???
Any ideas?

sagardubal, 2016-09-29:
We are using S7-1200 CPU, Able to read and write all the variables except for the Q and I, as such we are not getting any error. 

[image]

sembaye, 2017-03-13:
Fixed with the latest node.js version

closed 2017-03-13 by sembaye