plcpeople/nodeS7 #17: ET-200s Communication Problem

dedhdude, 2016-10-11:
Hi,

When I try to communicate with my "Et200s" PLC it is connecting with desire IP & Port Correctly, but then after it is Disconnected showing " Error - TCP Connected , ISO didn't

[image]

plcpeople, 2016-10-11:
Assuming it's an ET200S PLC and not an ET200SP PLC which would use slot 1...  

How many other connections are being made to the PLC?  Maybe it is being refused because there are too many connections made by other devices or reserved for other devices on the communication tab in the hardware configuration.  Will it work if you power cycle it and nothing else is connected?  Do other OPC servers/HMI packages work?

dedhdude, 2016-10-13:
Its is connected with SCADA.
And is working fine.

dedhdude, 2016-10-13:
My hardware Configuration
[image]

plcpeople, 2016-10-13:
What do you see if you look at the connection resources tab under CPU properties in TIA?  I have this same CPU for testing and it works fine.  If you have the web server enabled it will also give some information on what connections are in use.  I can't think of any other reason why it would deny the ISO connection other than there are no free connections.

plcpeople, 2016-10-22:
Did you have any luck with this?  I re-tested latest version with my ET200S without issue.

dedhdude, 2016-12-05:
Still I am Not able to connect with the plc.
Please Provide me your email ID so i can  send demo project.
Or you can mail me your demo project to Email Id:- "dedhdude@gmail.com"

dedhdude, 2016-12-10:
Did you checked the file which i had send u?
And i had tried with s7-1200 plc and is working good.

plcpeople, 2017-01-21:
This was solved via email so I am closing the issue.

closed 2017-01-21 by plcpeople

mstfldmr, 2018-12-11:
what was the problem?

plcpeople, 2018-12-11:
I looked back through my email and it was not clear.  A firmware update was the reported fix.  But I've connected with older firmware no problem.  If TCP connects but ISO doesn't, or other errors where a known address doesn't work, it's most likely that either all connections are reserved for other things (are you using TIA Portal or Step 7 - if TIA portal, check Device Configuration, Connection Resources and make sure the user-entered fields are all 0 or 1) or you're using a newer (1200/1500) PLC and haven't checked "Enable GET/PUT" in Device Configuration, Protection and Security, Connection Mechanisms, or you are trying to access an optimized data block.