plcpeople/nodeS7 #18: Write problem with bits

tomatensaft, 2016-10-27:
Hello,

at first of all - nodes7 is great !

The only problem i have is to write a bit. 
Conntection is OK and read items works fine - REAL, BITS, INT.
Write INT, REAL works also fine.

The only Problem is to write Bits

TEST:'M20.0',
TEST1: 'DB10,X4.0', 

For example i use
conn.writeItems('TEST', [true], valuesWritten);

The function valuesWritten bring me a "Done Writing" Message.

Do have a syntax mistake at my variables or at my code?

plcpeople, 2016-10-27:
What version are you using?  Version 0.1.10 was just added a few days ago and includes fixes for writing individual bits.  0.1.9 and earlier have issues with bits and bytes.

The same syntax you use to write REAL and INT should work for bits - conn.writeItems('TEST', true, valuesWritten)  should work fine, or conn.writeItems(['TEST'], [true], valuesWritten) - since TEST isn't an array, you want to write an individual value, not an array.

If it still doesn't work please let me know what the output is from conn.writeItems('TEST', true, valuesWritten).

tomatensaft, 2016-10-28:
Thx for you support - you're right.

It was my mistake!

I Used the valuecode for an array - conn.writeItems('TEST', [true], valuesWritten) 
Now i use the right One :-) ->
conn.writeItems('TEST', true, valuesWritten)  -> Now its working 

Thx

closed 2016-10-28 by tomatensaft