plcpeople/nodeS7 #20: two consecutive writing request

artexx, 2017-01-20:
Thank for you beatiful software.
I'm trying to simulate a pushbutton sending two consecutive request to write 1 and then 0 but there is a error message that I can't write if the previous activity is not completed. Evens using setTimeout with 3000 there is the same message. There is a way to solve?
Thanks
Walter

plcpeople, 2017-01-21:
Generally you want to put things like this in the callback for the write completion.  You also need to stop reading to do a write - but this logic should work without error for you. 

Like this:
    var ns7 = require('./nodeS7.js');
    var conn = new ns7;
    var doneReading = false;
    var doneWriting = false;

    conn.initiateConnection({port: 102, slot:1, host: '192.168.1.12'}, connected);

    function connected(err) {
         if (typeof(err) !== "undefined") {
        // We have an error.  Maybe the PLC is not reachable.  
            console.log(err);
            process.exit();
        }
        conn.setTranslationCB(tagLookup);
        conn.writeItems('BUTTON',true,doneWritingTrue);
    }

    function doneWritingTrue(anythingBad) {
        if (anythingBad) { 
            console.log("Error writing the button value to true"); 
            process.exit();
        }
        console.log("Done writing the button value to true.");
        setTimeout(conn.writeItems('BUTTON',false,doneWritingFalse),1000);
    }

    function doneWritingFalse(anythingBad) {
        if (anythingBad) { 
            console.log("Error writing the button value to false"); 
            process.exit();
        }
        console.log("Done with the button.");
        process.exit();
    }

    function tagLookup(tag) {
        switch (tag) {
        case 'BUTTON':
            return 'M50.0';
        default:
            return undefined;
        }
    }

Lichtjaeger, 2017-01-21:
I solved this using Promises:

```javascript
const nodes7 = require('nodes7')
const conn = new nodes7()

const variables = {
  bool1: 'DB1,X0.0',
  bool2: 'DB1,X0.1',
  int1: 'DB1,INT2'
}

conn.initiateConnection({
  port: 102,
  host: '192.168.0.2',
  rack: 0,
  slot: 1
}, connected)

function connected(err) {
  if (typeof (err) !== "undefined") {
    console.error(err)
    process.exit()
  }

  conn.setTranslationCB(tag => variables[tag])
  conn.addItems(Object.keys(variables))

  write({ bool1: true })
    .then(() => write({ bool1: false }))
    .then(() => read())
    .then(console.log)
    .catch(err => console.error('Something went wrong!'))
    // Sorry, but "err" is sometimes only a boolean
}

function write(object) {
  return new Promise((fulfill, reject) => {
    const keys = Object.keys(object)
    conn.writeItems(keys, keys.map(key => object[key]), err => {
      if (err) return reject(err)
      fulfill()
    })
  })
}

function read() {
  return new Promise((fulfill, reject) => {
    conn.readAllItems((err, values) => {
      if (err) return reject(err)
      fulfill(values)
    })
  })
}
```

artexx, 2017-01-21:
Thank you very much for the quick reply to both.
plcpeople solution worked smootly,  Lichtjaeger solution gives me the following error that I'm not able to solve because of my poor experience in javascript.
``
conn.setTranslationCB(tag => variables[tag])
SyntaxError: Unexpected token >
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
``
I'm using it in paralles with node-red-contrib-s7 in node-red for reading plc variables.
I'm having trouble when the number of variale read is bigger than 19, there are limit on buffer size or array of variable size?
Thanks again
Walter

plcpeople, 2017-01-21:
Lichtjaeger's solution is more modern as it uses ES6 and mine did not - it looks from the error like you need to upgrade your node.js install.  (Check your version with `node -v` from the command line, I'm not sure exactly what version you need but it's probably time to upgrade.)  Sites like es6-features.org and others can help understand the ES6 features in Lichtjaeger's code.

What kind of trouble are you having when the variables read are bigger than 19?  If you read more variables, nodeS7 will need to make more requests to the PLC to get all the variables, so it will take longer, which normally is not a problem but it can be a problem if you are relying on timeouts rather than promises or callbacks.  Applications that read hundreds of variables (or arrays with hundreds of elements) are common and should not be a problem unless there is a bug.  If nodeS7 is giving an error, if you post it here we might be able to help.

Lichtjaeger, 2017-01-22:
Arrow functions are supported [since Node v4](http://node.green/#arrow-functions).

artexx, 2017-01-22:
Thank you again for you support.
Indeed my node on raspberry was stuck on v 0.10.29 even if apt-get said that it was update to the latest version. So I manually replaced node to 6.9.4 and now Lichtjaeger example works perfectly.
I'm now moving the flow from raspberry to my laptop to try to understand where is the problem because some times I have more problem on reading values and sometimes not. Maybe the raspberry PI 1 is too limited in power or the plc is too slow in sending replay or it doesn't accept more than 2 or 3 connection request in parallel.
I'll send you a feed-back once I'll understand where is the problem
Walter

Lichtjaeger, 2017-01-24:
Yes, you need to update the apt registration if you want to change the major Node version.

The command is this: `curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -` (see [here](https://nodejs.org/en/download/package-manager/))

closed 2017-03-22 by plcpeople