plcpeople/nodeS7 #22: No right Answer when reading a String

sembaye, 2017-03-10:
Hello everyone,
i am not 100% shure if this is a bug. I have following issue:
Within my PLC (S7-300) i have a Datablock DB111. Address 0 is parameterized as STRING[12] with the content "SIMATIC--PLC".

When i read out following variables:
`var variable = [
	'DB111,CHAR0.14',
	'DB111,STRING0.11',
	'DB111,STRING1.12',
];`

i'll get:
`{ 'DB111,CHAR0.14': '\u0000\u0000SIMATIC--PLC',
  'DB111,STRING0.11': '',
  'DB111,STRING1.12': 'IMATIC--PLC\u0000' }`. 

Interpretation:
'DB111,CHAR0.14', =>should be alright, two bytes infront of the actual string.
'DB111,STRING0.11', => Wrong
'DB111,STRING1.12',=> For testing purpose

sembaye, 2017-03-10:
Sorry, my mistake. It is working as it should.

closed 2017-03-10 by sembaye

gfcittolin, 2017-03-10:
Just to make sure, as I've encountered a similar problem: The problem here was that the first two bytes should contain the String length that, in your case, were zero, right?

plcpeople, 2017-03-10:
When the string is properly formatted and declared as a string of length N, the first byte is the maximum number of characters in the string, and bytes 3 to N+2 are the characters.

When READING the string, nodeS7 ignores the first byte (it assumes this byte is not needed since the length of the string is specified in the translation/address definition) , and will read up to either the length of the string specified or the number of characters in the second byte, whichever is less. 
 
So for a string declared to start at DBB0 - DBB0 is the max length (ignored when reading) and DBB1 is the actual length (used when reading).

If you have a string that doesn't conform to this and you don't want to use a length byte, my suggestion is to use a character array - those don't use the prefix bytes.