plcpeople/nodeS7 #23: Some writing constallations aren't working

sembaye, 2017-03-14:
Hi,

I evaluated some writing constellations and saw that some are not working:

	`conn.writeItems('M0.0', [false], valuesWritten);	//OK
	conn.writeItems(['MB0.0'], [false], valuesWritten);	//**Not Working**
	conn.writeItems(['M0.0','M0.1','M0.2','M0.3'], [true,true,false,false], valuesWritten);//OK
	conn.writeItems('MB0', [7], valuesWritten);		//OK
	conn.writeItems(['MB0'], [6], valuesWritten);		//OK
	conn.writeItems('MB0.3', [ 1,2,3 ], valuesWritten);	//OK
	conn.writeItems(['MB0.3'], [ 2,3,4 ], valuesWritten);	//**Not Working**`

Is this on purpose??

plcpeople, 2017-03-14:
There are a couple of issues here.

Issue #1 MB0.0 is technically a byte array of length 0 which is not valid.  MB0.0 is not a valid bit address when entered that way within Step 7/TIA Portal so I don't think we should support this as a valid bit address here.

For the last item, ['MB0.3'] to [2,3,4], you are writing an array of bytes so you would need to use ['MB0.3'] to [[2,3,4]] for it to work.

We should add some type checking when writing.  This would fail more of your examples that aren't properly typed, and would still fail the two that currently don't work.

sembaye, 2017-03-15:
Ups, you right. The first one was an typo, sorry for that. Just tested it, works fine.
I understand. Thanks

closed 2017-03-15 by sembaye