plcpeople/nodeS7 #24: Trying to call conn.readAllItems inside setInterval

kochecc2, 2017-03-21:
Here is how I am writing my setInterval call:
```
var myTimer;
myTimer = setInterval(conn.readAllItems, 2000, function(err, values){
	if(!err){
		callback(null, values);
	}
	else{
		console.log(err);
		callback(err);
	}
})
```
The callback I reference is in the wrapper function where I am setting the interval. I tried passing it by placing "callback," after the 2000, and before the function(err, values)... but that didn't seem to change anything.

Here is what I get when I try to run it:
```
[27462038,628543252] Unable to read when not connected. Return bad values.
/home/ssi/nodemonitor/node_modules/nodes7/nodeS7.js:504
        if (self.isWaiting()) {
                 ^
TypeError: Object [object Object] has no method 'isWaiting'
```

If there is a better way to constantly monitor tags I'm all ears.

plcpeople, 2017-03-22:
My understanding of the reason for the error is because when you call setInterval this way, conn.readAllItems is called with the global scope - try wrapping it like so:

    myTimer = setInterval(function(){
        conn.readAllItems(function(err, values){
            if(!err){
 	    	    console.log(values);
	    } else {
	            console.log(err);
            }
    })
    },2000);

This will sort of work but I don't like this either.  I would prefer to see triggering a setTimeout WHEN THE PLC REPLIES (see also #20) rather than triggering on a setInterval.

kochecc2, 2017-03-22:
Thanks, that seems to have worked. Here is my new code.

```
conn.readAllItems(itemsRead)
		
function itemsRead(err, values){
	if(!err){
		callback(null, values)
		timeouts['F01'] = setTimeout(function(){
			conn.readAllItems(itemsRead)
		}, 2000)
	}
	else{
		conn.dropConnection()
		console.log(err)
		callback(err)
	}
}
```

I'm parsing the callback and sending updated values to websocket clients. When the last client disconnects, I am clearing the timeout. Should I also do conn.dropConnection() when that occurs?

plcpeople, 2017-03-22:
That looks better, I will close the issue.  

Leaving the connection open depends on the application.  We usually open the connection when our application starts and leave the connection open, and when the Node process exits, the TCP connection is dropped and the PLC frees up the connection resource, but your application may be different.  Ours is constantly polling and logging when active anyway, so we wouldn't want to call dropConnection.

closed 2017-03-22 by plcpeople

kochecc2, 2017-03-22:
The idea of my application is to only addItems for sections of the system that are being actively monitored by our websocket clients. I suppose as long as any area (predefined set of Tags) is being viewed, the connection should remain open.

That being said, when the last client logs out of a specific area (hard-coded here as 'F01') I should call conn.removeItems for every tag in that area. 

Thanks again!

Now to figure out how to split up the results when I callback with the values from more than one area...