plcpeople/nodeS7 #26: Issue when receiving multiple packets at once

gfcittolin, 2017-04-27:
We're having an issue, where nodeS7 is treating a packet that is apparently correct as invalid.

```
...
[997,136871469 172.25.24.210 S2] SendReadPacket called
[997,137073486 172.25.24.210 S2] Sending Read Packet
[997,137322621 172.25.24.210 S2] Sending Read Packet
[997,138683123] INVALID READ RESPONSE - DISCONNECTING
[997,138820445] TPKT Length From Header is 231 and RCV buffer length is 468 and COTP length is 2 and data[6] is 128
[997,138856085] <Buffer 03 00 00 e7 02 f0 80 32 03 00 00 00 02 00 02 00 d2 00 00 04 0b ff 04 00 70 00 00 00 00 00 09 00 00 00 00 00 00 00 00 ff 04 00 70 00 00 00 00 00 00 00 ... >
[997,139296666] ConnectionReset is happening
```

I've analyzed it a bit and, for the case we have here, it looks like that `onResponse` is being called with two packages in the buffer (maybe the system was a bit slow at the moment) - this explains the TPKT length being 231 and the buffer length being 468 for the example above.

The code even treat this special case [here](https://github.com/plcpeople/nodeS7/blob/master/nodeS7.js#L1092), but it looks like the code isn't reaching this anymore because of the fastACK check being mate by `checkRFCData`:

```javascript
   //...
   }else if((LastDataUnit >> 7) == 1 && TPKT_Length === data.length){
   //...
   }else{
      ret='error';
   }
```

this causes `ret` being assigned `'error'`instead of the buffer. Maybe the if-else comparison should be changed to:

```javascript
   //...
   }else if((LastDataUnit >> 7) == 1 && TPKT_Length <= data.length){
   //...
   }else{
      ret='error';
   }
```

thus allowing the check previous mentioned check to be executed and succeed.

I have no experience with this fastACK functionality, therefore I don't know if this could break something. But if you agree on that I can create a PR for the modification above

sembaye, 2017-04-27:
Hi,
which version are you using?

gfcittolin, 2017-04-27:
Hi,
this happened with 0.1.12. By the way, when reverting to 0.1.11 the problem
doesn't happen anymore

sembaye, 2017-04-27:
Hi,
i just had a look into the Buffer. As far as I can tell is that within the RFC Header the length of the whole RFC packet is defined as 231 bytes (see Byte 3 of fastACK packet). But for some reason the whole buffer is much higher (data.length = 468). I don't know if this is a mistake or not but the RFC 1006 protocol expects at that point 231 Bytes.

"My opinion" is that TPKT_Length and data.length has to be equal.

Can you analyze the communication with wireshark and the s7comm plugin. It would be interesting what kind of data are coming back from the plc.

plcpeople, 2017-04-28:
I spent a while on a similar problem when developing this and the similar Allen-Bradley module as well - the PLC sends out multiple request packets almost at once, and if the responses come in to node at essentially the same time, even if they are separate packets in Wireshark, sometimes node gets two (or possibly more) packets as a single buffer in the 'data' event.  So in this case TPKT_Length is way less than data.length, so we only process the first bytes (as specified in TPKT_Length) and treat the rest as a separate packet which in my earlier testing is processed correctly.  So if RFC expects 231 bytes then the correct thing to do is process the first 231 bytes, assuming it passes validation.  And you call onResponse again with the timeout, which will hopefully validate the next 468-231=237 bytes as another valid packet.  

I think the fix is correct, unfortunately I don't have anything that sends fast acknowledgement to test the fast ack code.  But it looks to me like it won't be affected by this fix.

sembaye, 2017-04-28:
I can test fastACK scenarios in my company (WinAC). I won't be there for the next three month but i can ask a collegue to test it. Or i'll do it by my own. Does not take much time.
Just send me the change and i'll verify it as quick as possible.

plcpeople, 2017-04-28:
I had another look at this and I think this is the correct fix, and won't affect fast ack, so if you want to submit a PR I will merge it.

gfcittolin, 2017-05-01:
Hi all,

just created the PR #28 for this. 

I was also able to simulate the problem by simulating some load that would block the event loop long enough, so that NodeJS would queue multiple packets on the buffer, just like the first log I've sent. I ran this with a server created with snap7, not a real PLC, although the results should be the same.

Here you can find the (very very crude) code I used to test, the output log, and the wireshark dump of it [fix-26_simulation.zip](https://github.com/plcpeople/nodeS7/files/968635/fix-26_simulation.zip)

@sembaye I tried to test the most I can here, but it would be perfect to see if it still works for you :)

plcpeople, 2017-05-01:
Thanks for this - I merged the PR and based on your testing will close the issue.  If any problems are seen with fast ack, we can work around it as we definitely need to be able to receive multiple packets in one buffer on slower machines.

closed 2017-05-01 by plcpeople