plcpeople/nodeS7 #32: Need examples

ceremcem, 2017-06-16:
It would be good to have example codes.

plcpeople, 2017-06-17:
What kind of example codes beyond what's in the README would you like to see?

ceremcem, 2017-06-17:
Having an `examples/` directory and examples like `simple-read.js`, `simple-write.js`, `dynamically-add-addresses.js`, ...etc in it would be great. A ready-to-run code is always preferred. 

It's OK to have an example code in the README (and I started using your library by this example code) but it was not running in the first place as I copied and pasted into editor, because there was a naming issue (`nodes7` should be changed to `nodeS7`) in the `require` line. In order to prevent these kind of typing errors, we should keep examples in a ready-to-run manner.

Let me create a PR as soon as I finished this code [in my application](https://github.com/aktos-io/aktos-dcs-node/tree/6c57c0f040eb4251a08dcbdb7b586e6d48ae31d4/protocols/siemens).

plcpeople, 2017-06-22:
I agree with the suggestion, this would be better.  Thanks - PRs are always appreciated.  And your application looks pretty cool...

I tried the example in README and it worked using npm install on both Linux and Windows as-is.  The project was developed as "NodeS7" and changed to nodes7 at the last minute, as required to publish on npm at the time.  I am not sure the best solution to the example not working when cloned and referenced directly.  Changing the file name to nodes7.js (from nodeS7) will likely solve the issue, but may then also create other issues for others, so I am not sure if it is a good idea at this point.   Any thoughts?

ceremcem, 2017-06-22:
Thank you. Your library powers the Siemens side of it :+1: 

The only proper way I can think of is changing name from `nodeS7` to `nodes7`. If you make a [new release](https://github.com/plcpeople/nodeS7/releases) (where you have none right now) and publish a **breaking change** section in the changelog of that release, everything would be good. 

As this library is beta (even alpha, as you mentioned), breaking changes are always expected.

plcpeople, 2017-06-22:
OK, I will consider doing that.  

Given the number of users and use cases of the library, I should probably remove at least the reference to alpha code.

ajitrishabh, 2018-07-03:
@plcpeople  is there any status on the example part?

ceremcem, 2018-07-03:
Nothing on my side currently. What part did you stuck on?

ajitrishabh, 2018-07-03:
Thanks for the quick reply.
i am very new to PLC world (yesterday only i first time saw PLC) and TCP/IP connection. i don't even know where the data get stored in Plc. one of my team member have some understanding on plc but he is on leave for 2 week.
im PHP developer with some nodejs experience. i have S7 1200 plc i just want to get values from plc and create some graph. from yesterday im looking for some solution. 
below is the output i got after running the nodes7 library.

`$ node index.js
[94984,367985877] Initiate Called - Connecting to PLC with address and parameters:
[94984,373914525] { port: 102, host: '192.168.0.1', rack: 0, slot: 1 }
[94984,378846374] Connection cleanup is happening
[94984,386617127 192.168.0.1 S1] Attempting to connect to host...
[94984,394912964 192.168.0.1 S1] TCP Connection Established to 192.168.0.1 on port 102
[94984,395419164 192.168.0.1 S1] Will attempt ISO-on-TCP connection
[94984,398566795 192.168.0.1 S1] Using rack [0] and slot [1]
[94984,405850234 192.168.0.1 S1] ISO-on-TCP Connection Confirm Packet Received
[94984,460565727 192.168.0.1 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
[94984,461795713] Translation OK` 

need to read values from plc. pls guide me on this.

Lichtjaeger, 2018-07-03:
To access values you need their addresses. They can be stored in data blocks (DB), in memory (M) or you can access the physical inputs (I) and outputs (O). You can get an address list from the engineer who programmed the PLC. Also, he needs to enable PUT/GET access to these blocks so you can reach them. (In old PLCs this is enabled by default)

ajitrishabh, 2018-07-03:
Thanks @Lichtjaeger .
is there any way that i can access address, data block(DB). as he is on leave i do have access of TIA portal.
if you can guide me on this.

Lichtjaeger, 2018-07-03:
- open the project in "Project view" mode
- the navigation should be on the left side
- expand the PLC you are looking for
- expand "Program blocks"
- search for the right DB (the blue cylinder icons)
- double-click it to open it
- there should be a column "Offset" (if there isn't one then PUT/GET access is not activated for this DB)
- press CTRL + B to make sure the addresses are up-to-date
- now you have a list you can start with

Lichtjaeger, 2018-07-03:
You can build the addresses the following way:

- first, you need the DBs number (e.g. `DB2`)
- then there is the Data Type (`DB2,DINT`)
- next comes the Offsets first number (`DB2,DINT20`)
- and last you can optionally give an amount of how many values you want from this type (`DB2,DINT20.2`) (generates an array)

The only exception is from type Bool. The addresses of booleans are a bit different.

paskalare, 2018-08-21:
Can you please give an example syntax of bool inside DataBlock?

babinc, 2018-08-23:
Just off the top of my head for a bool i think is it DB2,X20.0

babinc, 2018-08-23:
Here is some code that will convert a regular Siemens address to a NodeS7 address.

https://gist.github.com/babinc/7fc1f1cbb9ce533e74492661918e3d11#file-convert-js

paskalare, 2018-08-23:
you are correct!
thanks