plcpeople/nodeS7 #37: Will it work also with Siemens Logo 8?

galloxD, 2017-08-14:
Hey,

can somebody tell me if it work with a Logo 8?
I testet already but i can't change or read values.
They sad the Logo is online but nothing more.

Thx

plcpeople, 2017-08-14:
Yes, this was added thanks to a PR from gfcittolin and you have to make sure the TSAP is set properly in nodeS7 and also in your project.  This has been tested with the node-red-contrib-s7 project and is discussed there with some good screenshots - see https://github.com/netsmarttech/node-red-contrib-s7/issues/5

Please see also the addressing section at https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/LOGO.htm

In summary - use DB1,INT0 as an address for VM0.  Use DB1,INT1118 for AM1.  It looks like inputs, outputs and memory bits are offset by 1.  See the link for more examples.  

Hope this helps.

plcpeople, 2017-08-25:
I am going to close this as it is believed to be working. If not please let us know.

closed 2017-08-25 by plcpeople