plcpeople/nodeS7 #44: Packet Timeout due to Fase Acknowledge

sembaye, 2017-09-20:
Hi,
currently i have a problem that due to the lovely Fast Acknowledge a timeout occurs.

See:
[image]
[org.txt](https://github.com/plcpeople/nodeS7/files/1317366/org.txt)

Until the “onResponse” function, the communication is alright.

I would we should somehow cancel the timer or re-trigger it here:
https://github.com/plcpeople/nodeS7/blob/3f70a74e0609f09f32fad394e251dc34406e1486/nodeS7.js#L1088

I tried it out with “clearReadPacketTimeouts” but then further communication was not possible anymore. How can I restart the timer?

plcpeople, 2017-09-20:
There are potentially multiple timers, one for each packet that is sent.  So I don't think we necessarily want to re-trigger all timers on receiving a fast ack.  (Although it could be done.)  Does increasing self.globalTimeout (currently hard-coded on line 71) to higher than 1500ms help in your application?  If so, maybe allowing this timer to be adjusted when creating a new connection is a valid solution.  

I really wish we had an old PLC so I could test with fast ack, but we don't.

sembaye, 2017-09-21:
OK, thank you.
I have new knowledge about Fast Acknowledge and the entire RFC layer. As soon as I fixed my bug(s), I will inform you about it in the wiki.

ceremcem, 2017-09-21:
@sembaye IMHO, You may leave this issue open, as it is not meant to be solved yet. If any of us is unable to find an appropriate PLC and test it or solve the problem, @plcpeople would close it anyways.

sembaye, 2017-09-21:
oki doki