plcpeople/nodeS7 #47: ECONNRESET error with S7-400H

pstorch, 2017-11-07:
Is there any experience with S7-400H (it is the high availability version with two IP-Addresses)?
I get the "Error connecting to PLC: Error: read ECONNRESET" error.

I'm using it via node-red 0.17.5, node-red-contrib-s7 1.4.0 and nodes7 0.2.5.

plcpeople, 2017-11-07:
Unfortunately we can't afford one of these setups for testing :-) although I have used them in the past myself, but not with nodeS7.

Do you get this error all the time or only when switching processors or only when connecting to the PLC/CP that isn't the active one?

pstorch, 2017-11-07:
I get this error all the time. I tried both IP-Addresses.

plcpeople, 2017-11-07:
I have seen this error when connecting to the wrong slot number.  Are you changing the slot number of the PLC from 2 to something else when passing options to nodeS7?  What slot number is the actual PLC in?

Another thing that can cause this is there being not enough S7 connections available due to the connection configuration in Step 7, although I don't think this is the issue with an H system.

pstorch, 2017-11-08:
@plcpeople thanks for the hint with the slot number. I used 2 because I copied the configuration from a windows software (ServiceLab12). This uses CPU Slot 2 successfully. I don't know why, but I changed it to 0 and got data now. Still need to verify, if the data is correct, but I can continue from here.
Thanks again, I'll close here.

closed 2017-11-08 by pstorch