plcpeople/nodeS7 #48: Use questions

anthrogan, 2017-11-14:
Hi, just a few questions from me 😃 

I'm planning on including this in a restful API to avoid the need of a SCADA/OPC type 3rd party system and a database to pass information back and forth to a PLC. It's working up to now (Where by I have a basic api with express returning values from my datablock and posting any changes i make from a web front end)

Currently there are lots of outputs from the module, is there a way to turn these off 'in general' not error stuff just the normal information messages or will I have to do this myself? It's just so I don't fill up my logs with unnecessary information or will I have to go through one by one?

Also, instead of process.exit(); that is used in the example I call conn.dropConnection(); I'm concerned the connection is left open once a query has been made. Is this the intended use of this? I get connectionReset logs. If I don't drop the connection I can only make one get request and the ones after just hang, if I call dropConnection I can makes lots of requests in quick succession.

Any help appreciated!
Rogan.

plcpeople, 2017-11-16:
Hi...

To reduce the output, you can pass in ({silent: true}) to the constructor.  This may still not do what you want though, it may not show errors that you want to see.  If you do make major improvements to logging please submit a PR if you can, I realize there is room for improvement here.

It is more efficient to leave the connection open and make requests when you need the data, you'll get faster response that way without the delay of having to set up the connection each time and no other operator panel or software can take the connection once you've opened it (although this is less of an issue with newer PLCs).  If your second request for data is hanging, I would suggest you turn debugging on (pass in ({debug: true}) to the constructor) and post the log as an issue and we can take a look at what is going on as this is a very typical use case - open a connection, request data, and in the callback (with the data) you request data again, or wait a bit then request data again.

conn.dropConnection() will end the TCP connection without quitting Node, process.exit() quits Node which will drop the TCP connection on exit anyway.  conn.dropConnection() is correct if you are not quitting the process.  I should probably show this in the example/documentation.

anthrogan, 2017-11-17:
Hi

Thanks for the response. I'll give the logging a look over and send a PR if I make any changes.

The reason I ask for about the dropConnection is I won't be using the access for general logging/viewing of PLC variables like a normal SCADA/Web front end SCADA system. I'll be using it to update some variables we have for shift times. These are updated infrequently so "fast" response isn't necessarily a requirement, just the ability to write the DB areas at all is the idea here.

Much appreciated,
Rogan