plcpeople/nodeS7 #49: Nodes7- Siemens S7 200 Smart. Variable word memory Variable Byte VB not supporting

sushilbisne, 2017-11-25:
Hello,
I have tested this node with node-red with my raspberrypi and Siemens S7200 smart PLC
It supports variable like. Input (I0.0) , Output (Q0.0), Memory Variables MW, MB. and working well.

But not able to read & write Variable Memory VB, VW , Vbit, and timer e.g T100 , counters.

Please help to how to get control this variable with nodeS7.

plcpeople, 2017-11-25:
I don't have access to an S7-200 PLC with Ethernet so I have not been able to test any of these things.  But I think you can access the V area using DB1.  

So to access V0.0, use DB1,X0.0 and to access VW8 use DB1,INT8 - please let me know if this works for you, other data types should work as well.  

S7-200 timers and counters have their own unique data type in the S7 protocol and support hasn't been added for those yet so this will not work.  I would consider adding in the future if you would be able to test but hopefully the V area will get you going for now.

sushilbisne, 2017-11-26:
Hello,

Thanks ,, Its working .. Now problem with Timers & Counter register only.

Please update once you update this node for the same..

and thanks once again..

plcpeople, 2018-06-17:
Closing this issue as support for V memory exists through DB1.  Lack of support for S7-200 timers and counters moved to issue #58.

closed 2018-06-17 by plcpeople