plcpeople/nodeS7 #50: use nodeS7 with browserify

timo-plc, 2018-01-25:
hey,
I'm still a beginner in programming but have become accustomed to it

I would like to use nodes7 to connect a website with a plc.

If someone has a better Idea I am happy about every suggestion.

what I've done so far:
Installation of node, browserify and the modules
create a file with browserify and integrate it into my website
```
We have an error.  Maybe the PLC is not reachable.Cannot open TCP connection [404]: <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL /api/vm/net/connect was not found on this server.</p>
<hr>
<address>Apache/2.4.18 (Ubuntu) Server at 192.168.2.124 Port 80</address>
</body></html>

Maybe someone can help or give me a hint

plcpeople, 2018-01-25:
The problem with browserify is there are a lot of things that will run on node.js that can't be run in a browser due to limitations of the browser in general, and this is one of them.  NodeS7 creates a TCP connection to the PLC using the node.js net module, and this can't be done in the browser at all as the API does not exist for it.  

To connect a website with a PLC using nodeS7 (or any other package I am aware of) you still need to write a basic server application in node.js, and then pass PLC data using socket.io or some other type of polling, unless you just want the value to update every time the page loads.

I would suggest starting from a simple example like https://socket.io/get-started/chat/ and modifying to include nodeS7 and to send data.

closed 2018-01-25 by plcpeople

timo-plc, 2018-01-25:
Thanks for your Help and your tips