plcpeople/nodeS7 #54: Can not connect to S7-1200 PLC

volkanalkilic, 2018-05-30:
Hi,
I can not connect to the S7-1200 PLC. However, I can connect to the same PLC using this library: https://github.com/killnine/s7netplus

What causes this problem?

Lichtjaeger, 2018-05-30:
Can you post your configuration? For S7 1x00 it should look like:
```javascript
const plc = new (require('nodes7'))({ silent: true })
plc.initiateConnection({
  host: '<IP of the plc>',
  port: 102,
  slot: 1,
  rack: 0
}, err => {
  // ...
})
```
Can you ping the plc?

Lichtjaeger, 2018-05-30:
Hmm..., should work.

Can you copy the console output? Is there an error?

volkanalkilic, 2018-05-30:
I do not have access to PLC at this point, but as far as I can remember console output is "internal server error".

Lichtjaeger, 2018-05-30:
The console output of nodeS7 looks like:

```
[919,883304001] Initiate Called - Connecting to PLC with address and parameters:
[919,884208285] { port: 102, host: '192.168.1.150', rack: 0, slot: 1 }
[919,886065128] Connection cleanup is happening
[919,900610458 10.2.202.145 S2] Attempting to connect to host...
[919,908272737 10.2.202.145 S2] TCP Connection Established to 10.2.202.145 on port 102
[919,908350279 10.2.202.145 S2] Will attempt ISO-on-TCP connection
[919,913547538 10.2.202.145 S2] ISO-on-TCP Connection Confirm Packet Received
[919,918113432 10.2.202.145 S2] Received PDU Response - Proceeding with PDU 240 and 2 max parallel connections.
```

If you get an error before this lines, your program is crashing before even reach your `conn.initiateConnection` line.

plcpeople, 2018-06-17:
Any updates on an actual connection error?

plcpeople, 2018-07-11:
Closing due to lack of activity.  Please re-open if you are still having difficulty and can post some console output or other information that might help resolve the issue.

closed 2018-07-11 by plcpeople