plcpeople/nodeS7 #55: Cannot access some addresses on S7 1200

semihtezcan, 2018-06-05:
Hi,
We are having trouble reading from some addresses.

[image]

For the addresses above, we can always read from and write to "MW100" whereas we cannot read from or write to "DB3.DBD0" using same code. When we try to read, the console says "Failed to find a match for DB3".

What are we doing wrong?

plcpeople, 2018-06-05:
Try DB3,REAL0 - the syntax is slightly different as you are specifying the data type as well, as NodeS7 can't determine that any other way.  The data block has to be non-optimized as well, although it looks like this is already the case from your picture.

plcpeople, 2018-06-17:
Did the suggested address change work for you?

plcpeople, 2018-07-11:
Closing due to lack of activity.

closed 2018-07-11 by plcpeople