plcpeople/nodeS7 #67: Cannot Read DB155.DBD332,FLOAT type on S7 1200

ptertrung, 2018-09-24:
Hi plcpeople
I tried read data from PLC S7 1200. But I can't read data with data type float from PLC.
DB155.DBD332,FLOAT0
DB155.DBD332,FLOAT0.0
with no success.

could you tell me what the type of float is?

Lichtjaeger, 2018-09-24:
This is a list of the types and their aliases that are implemented in this library:
```
REAL
DWORD
DINT
INT
WORD
X
B
BYTE
S
STRING
C
CHAR
TIMER
COUNTER
```

There is no FLOAT. You are looking for REAL.

plcpeople, 2018-10-12:
I am going to close the issue.  Thanks for pointing out that REAL is the correct syntax.  DB155,REAL332 is a single floating point at offset 332.  DB155,REAL332.10 is an array of 10 values starting at this same offset.  Very similar examples are in the README.

closed 2018-10-12 by plcpeople