plcpeople/nodeS7 #69: Node-RED crashes while reading data

pabaxx, 2018-10-12:
Hello, i'm working on a student project and got a 'big' issue : 

i have a local s7-1500 i'm working with and a gateway with node-red container installed inside docker.

reading out the data with S7in Node is no problem, **but** when i'm using the example code (from the readme) and put in my adress (just an array of 8 integers for testing) ,  node-red crashes after ~120s.
First it works fine, i'm receiving the data as expected, but after this time i'm loosing connection to node-red. 
The problem is reconstructable..  any ideas?
best regards

plcpeople, 2018-10-12:
Can you please post some sort of stack trace, log or other information so that we know where in the code it's crashing?  It may not even be crashing in NodeS7.

plcpeople, 2018-10-12:
Are you able to get a stack trace or log from that new container?

pabaxx, 2018-10-12:
i'm very new to this all, but i installed stacktrace-js and got this output, no idea wether its helpful or not..

> Function()@node:eeea8b0d.93ffa8:13:12
Function()@node:eeea8b0d.93ffa8:17:3
ContextifyScript.Script.runInContext()@vm.js:59:29
FunctionNode.<anonymous>()@/usr/lib/node_modules/node-red/nodes/core/core/80-function.js:224:33
emitOne()@events.js:116:13
FunctionNode.emit()@events.js:211:7
FunctionNode.Node.receive()@/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:215:14
InjectNode.Node.send()@/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:202:14

plcpeople, 2018-10-12:
By stack trace I mean the console output when the node process crashes, if it's actually crashing rather than just stopping communicating with the PLC.  Can you access the console output through the container somehow?  Docker logs maybe?

pabaxx, 2018-10-15:
unfortunately, i cannot access the docker logs because the host system is secured and i only have access to the docker UI. 
The communication doesnt stop instantly, first i get the same outputs several times and then the connection to node-red gets lost..

Lichtjaeger, 2018-10-15:
Can't you run your program locally instead of the container for debugging purposes?

pabaxx, 2018-10-15:
I just saw that docker gives me an error, maybe you can tell me something about it..

> DPM status changed 0xFF (OK => 0x11)!
Error SPIReadChunk: DPM Addr=0x200 / Len=0x4

plcpeople, 2018-10-15:
That is not the information we're after.

Please let us know if you can duplicate the issue locally.  Do you have local access to a PLC?  If so, install Node-RED and everything locally, in a Docker container or just on its own, and see if you can duplicate the problem that way.

pabaxx, 2018-10-16:
&this is everything i'm using

> var nodes7 = global.get('nodes7');  // This is the package name, if the repository is cloned you may need to require 'nodeS7' with uppercase S
var conn = new nodes7();
var doneReading = false;
var doneWriting = false;

var variables = { TEST1: 'DB3,INT2.8' 		// Memory real at MD4
//		  TEST2: 'M32.2', 		// Bit at M32.2
//		  TEST3: 'M20.0', 		// Bit at M20.0
//		  TEST4: 'DB1,REAL0.20',	// Array of 20 values in DB1
//		  TEST5: 'DB1,REAL4',		// Single real value
//		  TEST6: 'DB1,REAL8',		// Another single real value
//		  TEST7: 'DB1,INT12.2'		// Two integer value array
};

conn.initiateConnection({port: 102, host: '192.168.222.5', rack: 0, slot: 1}, connected); // slot 2 for 300/400, slot 1 for 1200/1500
//conn.initiateConnection({port: 102, host: '192.168.0.2', localTSAP: 0x0100, remoteTSAP: 0x0200, timeout: 8000}, connected); // local and remote TSAP can also be directly specified instead.  The timeout option specifies the TCP timeout.

function connected(err) {
	if (typeof(err) !== "undefined") {
		// We have an error.  Maybe the PLC is not reachable.
		console.log(err);
		process.exit();
	}
	conn.setTranslationCB(function(tag) {return variables[tag];}); 	// This sets the "translation" to allow us to work with object names
	conn.addItems(['TEST1']);
//	conn.addItems('TEST6');
//	conn.removeItems(['TEST2', 'TEST3']);  // We could do this.
//	conn.writeItems(['TEST5', 'TEST6'], [ 867.5309, 9 ], valuesWritten);  // You can write an array of items as well.
//	conn.writeItems('TEST7', [ 666, 777 ], valuesWritten);  // You can write a single array item too.
	conn.readAllItems(valuesReady);
}

function valuesReady(anythingBad, values) {
	if (anythingBad) { console.log("SOMETHING WENT WRONG READING VALUES!!!!"); }
//	node.error(values.TEST1);
    flow.set("values",values.TEST1);
	doneReading = true;
	if (doneWriting) { process.exit(); }
}

function valuesWritten(anythingBad) {
	if (anythingBad) { console.log("SOMETHING WENT WRONG WRITING VALUES!!!!"); }
	console.log("Done writing.");
	doneWriting = true;
	if (doneReading) { process.exit(); }
}
msg.payload = flow.get("values");
return msg;

gfcittolin, 2018-10-16:
Are you putting this code inside a function node on Node-RED? Please note that the whole code is executed on every incoming message. This means you're creating a _new_ connection on every message, _without_ cleaning up the old one. So you end up using all the connection resources of the PLC and it won't accept new connections, that's why you're getting the `ECONNRESET`.

Why don't you use the [node-red-contrib-s7](https://flows.nodered.org/node/node-red-contrib-s7) node? It uses this library and handles all the details for you.

pabaxx, 2018-10-16:
Okey thank you, then thats the problem..
I wont use contrib-s7 because the DB addresses are variable and read out from another document..
I know i can upload an xml file but that should be automated so I'm using the code in a function for easy handling

pabaxx, 2018-10-16:
could i simply add dropConnection(doneReading) to the end of my function?

gfcittolin, 2018-10-29:
Adding `dropConnection()` may do it there. But take care not run  

Another (maybe cleaner) solution in your case would be to create a separate executable, where you can pass the variables over command line arguments or stdin, and then call it using the "exec" node, so you don't have this giant function in Node-RED.

closed 2018-11-08 by pabaxx