plcpeople/nodeS7 #72: not able to read or write from PLC s7-1200

ziborg99a, 2018-11-04:
I need help with read/write on PLC s7-1200.
I remember having same issues on NodeRed some time ago.
Is my PLC is faulty ??
Here is a log.

814827,224150445] Initiate Called - Connecting to PLC with address and paramete
rs:
[814827,229104172] { port: 102, host: '192.168.0.3', rack: 0, slot: 1 }
[814827,233475175] Connection cleanup is happening
[814827,237964775 192.168.0.3 S1] Attempting to connect to host...
[814827,381915361 192.168.0.3 S1] TCP Connection Established to 192.168.0.3 on p
ort 102
[814827,382169661 192.168.0.3 S1] Will attempt ISO-on-TCP connection
[814827,383903580 192.168.0.3 S1] Using rack [0] and slot [1]
[814827,388375504 192.168.0.3 S1] ISO-on-TCP Connection Confirm Packet Received
[814827,437803870 192.168.0.3 S1] Received PDU Response - Proceeding with PDU 24
0 and 3 max parallel connections.
[814827,438195584] Translation OK
[814827,439564587 192.168.0.3 S1] Adding TEST1
[814827,450946528] PDU type (byte 8) was returned as 2 where the response PDU of
 3 was expected.
[814827,451163196] Maybe you are requesting more than 240 bytes of data in a pac
ket?
[814827,451279513] <Buffer 03 00 00 13 02 f0 80 32 02 00 00 00 02 00 00 00 00 81
 04>
[814827,451825746] ConnectionReset is happening
[814828,949983208 192.168.0.3 S1] READ TIMEOUT on sequence number 2
[814828,951408659] Processing an undefined packet, likely due to timeout error
[814828,951912128] Timeout error - zero length packet
[814828,952043269 192.168.0.3 S1] Received a ZERO RESPONSE Processing Read Packe
t due to unrecoverable packet error
SOMETHING WENT WRONG READING VALUES!!!!
{ TEST1: 'BAD 255' }
[814828,955403913] ResetNOW is happening
[814828,971086153] We Caught a read/write error ECONNRESET - will DISCONNECT and
 attempt to reconnect.
[814828,971333612] ConnectionReset is happening
[814828,971749843] ConnectionReset is happening
[814830,472651584] ResetNOW is happening
[814830,472985139] Clearing an earlier scheduled reset

gfcittolin, 2018-11-05:
It looks like the PLC didn't like what you requested to it (the `81 04` in the end means "Context is not supported"). Make sure that:
 - You have disabled the "Optimized block access" for the DB you're trying to read (if it's a DB at all)
 - You have enabled the "Enable PUT/GET Access" (in the hardware config of the PLC)
 - You are trying to read a valid address

ziborg99a, 2018-11-09:
It works. !!! Many thanks for your help. 
Tested in Node-Red also with success.
Actually I was straggling to find this "Enable PUT/GET Access" option in  TIA portal, I had to fully open each tab in hardware config.
It was in the very bottom of one of them. :-(