plcpeople/nodeS7 #75: Help Polling in Read value

guleus, 2019-03-13:
Hi,
I'm very new to the Siemens world and also to node JS.
What I want to do is to create sort of an app/web app, where the users phone vibrates and rings when a certain variable on a siemens plc goes to 1.
So i found this library with which now I was able to read a value from the siemens plc.
What I don't know is how i make it check this value continuesly. In the library I found

this part in the readAlItems function

	if (self.isWaiting()) {
		outputLog("Waiting to read for all R/W operations to complete.  Will re-trigger readAllItems in 100ms.", 0, self.connectionID);
		clearTimeout(self.rereadTimer);
		self.rereadTimer = setTimeout(function() {
			self.rereadTimer = undefined; //already fired, can safely discard
			self.readAllItems.apply(self, arguments);
		}, 100, arg);
		return;
	}

but it never goes through here. 
Can somebody help this newbie please???
And where can i make such a note js run? Even on the Web Server of the Siemens itself???

Thanks a lot

plcpeople, 2019-03-13:
Why don't you start by running nodeJS on your computer and getting the example application in the README working?  The next step towards continuous polling would be to call readAllItems from the callback returning previous values.  The code snippet you mention above is not really intended for your use case.

guleus, 2019-03-13:
Hey thenks for the quick reply.
I already managed to to make it working on my pc.
The next step towards continuous polling would be to call readAllItems from the callback returning previous values--> sorry bit Im really new and don#t know how to do this

guleus, 2019-03-13:
Well I did this now:
  function valuesReady(anythingBad, values) {
  	if (anythingBad) { console.log("SOMETHING WENT WRONG READING VALUES!!!!"); }
  	console.log(values);
    conn.readAllItems(valuesReady);
  	doneReading = true;
  	if (doneWriting) { process.exit(); }
  }

and it reads but how can i access the value itself now? variable name 'Test2' is not working. If variable is true then do something

plcpeople, 2019-03-13:
What isn't working?  What have you done to try to get it working?

guleus, 2019-03-14:
I'm checking this value in the plc
  var variables = {
  		  TEST2: 'M32.2' 		// Bit at M32.2  		  
  };

But i don't know how to work with it (Acces the value itself). I would like to make an if statement and say, if  value is true then...

guleus, 2019-03-14:
Ok now I found out how to acces the value..   values["TEST2"]
So finally back to where I weant to get:
What I want to do is to create sort of an app/web app, where the users phone vibrates and rings when a certain variable on a siemens plc goes to 1.
Can I make this node js run on Web Server of the Siemens itself??
Or what suggestion would you guzys have.
Thanks,
Hannes

plcpeople, 2019-03-14:
NodeJS will not run on the web server of the PLC.  

There are a lot of very different ways to do what you're trying to do, but the actual communication with the PLC is probably less than 1% of the work and you have that figured out now, so this is not really the place to ask about the rest of your project.  Picking a framework or code sample for the app itself is probably your next step.

gfcittolin, 2019-03-14:
@guleus check [Node-RED](https://nodered.org), from there you can connect to your PLC to get the values cyclically (it uses this same library) and create a flow that could either serve a webapp of be the backend of a custom app of your own

armgit5, 2019-08-08:
I was wondering if we can have web socket realtime communication to s7 1200... for example if M4.0 value changes in plc, it will change on nodejs side as well without having to poll to read the data every 1 second also...

I was also curious what would be the difference between snap7 vs profinet vs ethercat vs powerlink protocol.... thank you.

Lichtjaeger, 2019-08-08:
@armgit5 For a more "real-time" like application I would recommend using a Modbus server instance in your NodeJS program and actively send values on change from the PLC side.

armgit5, 2019-08-08:
@Lichtjaeger thank you so much for your reply. is there any nodes7 command I can use that will listen to Modbus server value changes... 

I also saw a snap7 server library, I'm not sure if that could be something I could look at as well... http://snap7.sourceforge.net/

I was wondering if snap7 protocol is different than profinet... is profinet and ethercat a real time like web socket communication protocol? thank you

plcpeople, 2019-09-07:
I am just cleaning up issues so I am going to close this one, but as far as I know snap7 uses the same S7 protocol that nodeS7 uses.  PROFINET is used for IO communication and although nodeS7 and snap7 can get data from the "PROFINET interface" of a CPU they still use the S7 protocol.  NodeS7 doesn't do Modbus so if you choose to use that you are best to discuss that elsewhere.  I am not familiar with ethercat.

closed 2019-09-07 by plcpeople

armgit5, 2019-09-09:
Thank you so much for your feedback and explanation @plcpeople