plcpeople/nodeS7 #76: Concurrent readAllItems requests

KamalAman, 2019-03-20:
Two simultaneous requests to readAllItems should currently resolve their callbacks. 

However since nodes7 uses the contextual `readDoneCallback`, the second call to readAllItems interferes with the response with the first request. This results in the first requests callback to never get called, and the second requests callback to get called twice. 

```javascript
client.readAllItems((err, values) => {
  console.log(1);
});

client.readAllItems((err, values) => {
  console.log(2);
});

// 2
// 2
```

gfcittolin, 2019-04-12:
Indeed. Unfortunately, the library currently can't handle requests like this, and therefore you'd need to synchronize on the outside with something like (please note this hasn't been tested):
```js
var queue = [];
var currentCb = null;

function _processQueue() {
  if (currentCb) return; //there's a request in progress, skip
  currentCb = queue.splice(0,1);
  if (!currentCb) return; //there's nothing to request, skip
  client.readAllItems((err, values) => {
    currentCb(err, values);
    currentCb = null;
    _processQueue();
  });
}

function asyncReadAllItems(cb) {
  queue.push(cb);
  _processQueue():
}
```

There's a rework of the library going on that will fix this and other issues. So maybe we'll have to workaround until that work is done :)

LiaraT, 2021-04-20:
Hello. Could you write a slightly more detailed example of how to work with two data queries? I'm trying to make 2 requests for different data (twice I write different data to "conn. addItems", and twice I try to read "readAllItems"). As we wrote above, only the second request is executed.

I apologize for the mistakes, English is not my native language, so I use a translator.
Thank you in advance