plcpeople/nodeS7 #77: Profinet devices

lpmwfx, 2019-05-06:
Can the lib be used for communication with Profinet devices as frequency convertsers and remote IO's like ET200 from Siemens?

ceremcem, 2019-05-06:
Nearly Impossible AFAIK. Profinet and Profibus are proprietary protocols that requires an extensive reverse engineering effort.

lpmwfx, 2019-05-06:
Yes I thought so. But is that also true if I use 3rd party devices like a ASI gateway from IFM - They gladly are going to reveal the internal adresses.

ceremcem, 2019-05-06:
I don't know anything about the mentioned device, but any company can freely produce a proxy device. They can also mimic Profibus/net API for sure, which would give the impression of revealing the Profibus internals.

closed 2019-09-07 by plcpeople