plcpeople/nodeS7 #82: readall no return value

celevra, 2019-06-14:
Hi,

is it also possible to communicate with Tia Portal 1500?

regards

Celevra

plcpeople, 2019-06-14:
Do you mean with an S7-1500 PLC programmed with any version of TIA Portal?  Yes, as long as you use slot 1 in nodeS7, as long as you have GET/PUT enabled in the PLC configurations (I think it's under security, connection mechanisms) and as long as you use data blocks where "Optimized" has been turned off.  (Right click the data block and go to attributes)

Blume987, 2019-07-11:
Is there anything else to configure? I got an project for a S7-1500 PLC and started the simulation of the CPU. But I can't get a successful connection:

`[11819,461439735] Initiate Called - Connecting to PLC with address and parameters:
[11819,478408411] { host: '127.0.0.1', port: 102, rack: 0, slot: 1 }
[11819,481145051] Connection cleanup is happening
[11819,485210458 127.0.0.1 S1] Attempting to connect to host...
[11820,486824089 127.0.0.1 S1] We Caught a connect error ECONNREFUSED
[S7] Could not connect to 127.0.0.1:102 on rack:0, slot:1`

Lichtjaeger, 2019-07-11:
Did you enable PUT/GET access as @plcpeople stated?

And IP 127.0.0.1 will never be your S7-1500. That's the loopback address of your computer.

Blume987, 2019-07-11:
Yes, PUT/GET is enabled. I thought the simulator uses localhost. Where do I get the IP of the simulated PLC?

plcpeople, 2019-07-11:
PLCSIM doesn't accept external S7 connections - you can't use other 3rd party HMI packages with PLCSIM either.  For that you would need PLCSIM Advanced.  

You are seeing the TCP port being available, but PLCSIM doesn't accept the S7 connection once the TCP connection is established.

Blume987, 2019-07-11:
I read about the tool [NetSimPLC](http://nettoplcsim.sourceforge.net/). It publishes the internal simulation IP to connect from an external computer. Do you know this tool? Would it help in our use case?

plcpeople, 2019-07-11:
I used this tool years ago and it worked well for testing - I thought it only worked with Step 7 <= 5.5 and S7-300/400 but reading now, it looks like it works with TIA and a simulated S7-1200/1500.  It looks like you might want to try a port other than 102 or start the NetToPLCSIM application first to get it to claim port 102 before anything else does.  Please let us know how it goes, I may try this too as it would be useful if it works.

Blume987, 2019-07-12:
It worked, thanks a lot for your help!

Steps to get this working:

1. Open [NetSimPLC](http://nettoplcsim.sourceforge.net/)
1. Start TIA Portal and run simulation
1. Add PLC to NetSim
  * Use 127.0.0.1 as network IP
  * Detect PLCsim IP on clicking the (...) button
  * Select same rack and slot as in your code
1. Start NetSim server
1. Start your node application

plcpeople, 2019-07-15:
Thanks for the update and procedure for others.  I will close the issue now since it seems resolved, but if anyone has anything to add about getting nodeS7 working with PLCSIM Advanced, the info is welcome.

closed 2019-07-15 by plcpeople