plcpeople/nodeS7 #85: Symbolic

drexxi, 2019-08-02:
Hello,

is it possible to ask for a value with the symbolic adress?

Lichtjaeger, 2019-08-02:
NodeS7 uses the classic S7 protocol. This leads to the following disadvantages:
- not secure (must be switched on manually (PUT/GET enabling))
- only non-optimized blocks are accessible
- absolute memory addressing only

drexxi, 2019-08-02:
Hey thank u for your fast answer.
Now i have plcsim, nettoplc and im able to read my tags.

I want to start the read function from javascript in my browser.
I need to implement the code in a svg file, in the svg file im able to run javascript. 

do u think thats possible?

Blume987, 2019-08-07:
Hey drexxi,

as answered in this issue #50 it is not possible to run this library in browser.
I would also suggest to build your code on NodeJS and publish the data to browser via websockets.

closed 2019-09-07 by plcpeople