plcpeople/nodeS7 #92: Cant find the nodeS7 Blocks after installing

Performanzezero, 2020-01-11:
Hello 
this is the first time to write at GitHub community
i have installed Node-red at my Laptop to link it to S7-300 PLC
but after installing the module of S7, I can't find it as odes blocks at the left panel 
i have reboot the Node Services serval time, laptop and still can't find it 

Thanks for supports

plcpeople, 2020-01-11:
To work with node-red, you will want to install and follow the instructions for the node-red-contrib-s7 package.  This package is where the blocks you are looking for come from, and it also installs the nodeS7 package as a dependency for PLC communication.  You would not normally want to install the nodeS7 package on its own to work with node-red.  The github page for that repository has some documentation that should help you as well.

Performanzezero, 2020-01-11:
many thanks for express responding.. I have successfully installed it today and working to connect my PLC to node red

plcpeople, 2020-01-11:
Good to hear it's working...

closed 2020-01-11 by plcpeople