plcpeople/nodeS7 #95: Provide an example how to use this library

nagvenk, 2020-02-04:
Can you please provide an example how to us eth is library

ceremcem, 2020-02-04:
Did you see [README/Examples](https://github.com/plcpeople/nodeS7#examples)?

ventr1x, 2020-04-29:
Many things are not documented and simply not tested at all.
For example a simple dropConnection() (which aparently is not used by the maintainer) is most of the time not enough and messes up the plugin entirely. I had to use connectionReset() and resetNow() on a simple Siemens plc to even disconnect correctly (it would block the plugin otherwise for whatever reason).
These functions are not documented. Any exposed function should be mentioned.

nagvenk, 2021-01-11:
i am not working anymore on this

closed 2021-01-11 by nagvenk