plcpeople/nodeS7 #98: Emit event when PLC is not Reachable!

flacohenao, 2020-04-28:
On the branch **next**, would be nice If an attempt of connection fails due to unreachable PLC would emit some event indicating that.

I have 'error', 'connect', 'disconnect' but non of them emits anything on that case.

I was diving in the code and saw that you have a : "_onConnectionError" that emits the error on the "error" event but just when user does that? I think?

flacohenao, 2020-04-29:
Actually I was diving a little bit more... and I was using one of the (_) methods... and when I used the constructor... the 'disconnect' event did throw something.. but just a pulse... the error itself is not being catched

gfcittolin, 2020-04-30:
I've tested with the following snippet:
```js
const { S7Endpoint } = require('nodes7');

console.log('Connecting to PLC...')
let plc = new S7Endpoint({ host: '192.168.15.123', slot: 1 });

plc.on('connect', () => console.log('PLC#connect'));
plc.on('error', e => console.log('PLC#error', e));
plc.on('disconnect', () => console.log('PLC#disconnect'));
```

... which led me to the following output:
```
Connecting to PLC...
PLC#disconnect
PLC#disconnect
PLC#error { Error: connect EHOSTUNREACH 192.168.15.123:102
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
  errno: 'EHOSTUNREACH',
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.15.123',
  port: 102 }
```

You can see that it does emit an error, with `code` and `errno` set to `EHOSTUNREACH`.

Actually, I found out another bug: Even though it should retry automatically (`autoReconnect` in the default value, 5000 ms), it just didn't and the process exited. I'll open another issue for that.

@flacohenao would you mind sharing a snippet of code that reproduces the behavior you mentioned?

flacohenao, 2020-04-30:
@gfcittolin I'll do it as soon as I can.

Also, please see the #101 that i thing is the same issue you mention!

gfcittolin, 2020-04-30:
Just note that the error has to be emitted by the OS in the first place... just tested this here with a fully random IP, and does not happen (at least not immediately), because a connection request is sent out to the network's gateway and probably gets lost (or you get an ICMP error in some situations, mostly in closed networks). So it takes a really long time for the OS to decide the connection is "broken", and therefore we keep there waiting as well

Not sure if this is the behavior you're experiencing, but this can happen as well

flacohenao, 2020-05-02:
I think I need to test it when I can finally have access to physical PLC.

But what would be really really nice its having the augmented version of errors mentioned on the #101  when erros occurs

flacohenao, 2020-05-08:
Closing this issue, It was due to the Simulator!
Let's discuss the things about augmentng the errors on #110

closed 2020-05-08 by flacohenao