plcpeople/nodeS7 #100: Define the CONNECTED status and maybe provide proxy to get status in readable format.

flacohenao, 2020-04-29:
@gfcittolin 

Right now, we have ths defined status:

[image]

would be nice if you guys declare the 4 (CONNECTED) status in the source code, and maybe.. if you think is usefull (I think it is) provide a method to return the status not exposing the control property 

> conn._connectionState

but exposing a getter instead of be able to actually set a value through that property like:

> .getConnectionStatus()

that returns the "String version" of the states!

if not, we can close the issue.

gfcittolin, 2020-04-30:
I think what you mean by "READY" would be the `CONN_CONNECTED` state, right? If not, what would be the difference between them?

Note that there's already a getter called `isConnected`, so you could for example test it for `true` before trying to call other methods (or listening to the `connect` and `disconnect` events)

flacohenao, 2020-05-02:
Ok, nice!

I haven't seen the `isConnected` getter. didn't dive on source that much I gues...

closed 2020-05-02 by flacohenao