plcpeople/nodeS7 #99: Provide Handy Methods for Standar PLC Info!

flacohenao, 2020-04-29:
@gfcittolin 

So far, reading the main SZL lists is implemented, but I think it would be nicer if we Also expose the most common information of the PLC not just the byte buffer.

On my opinion, the most usefull information could be:

Get the CPU order code and version info, in wich we can encapsulate it on a function (I think the adress of this info is the same on each type of PLC)

Get the CPU info like: (ModuleTypeName), (SerialNumber),(ASName),(Copyright),(ModuleName)
that could be also encapsulate on a method or the same info method.

And.. Get the CP info like:  (MaxPduLength), (MaxConnections), (MaxMpiRate), (MaxBusRate)

I think that would be awesome! If it's impossible, then I guess it would also be acceptable that get that info depends on the user through the already developed getAvailableSSL() and get list by list.

gfcittolin, 2020-04-30:
I do agree that this would be handy, I'm just afraid of opening the doors to a function hell :) There are a dozens of SSL on the wild, with many different possible answers, and it would be very painful to maintain something that needs tons of if-elses to handle all corner cases . And even the most common ones may not be implemented in all devices, as far as I know.

I think it's worthy to implement these two: Module Identification (0x0011) and Component Identification (0x001c). At least we can start with these two, and maybe add others in the future.

flacohenao, 2020-05-04:
Very well implemented! Thank you very much!

closed 2020-05-04 by flacohenao