plcpeople/nodeS7 #114: S7ItemGroup.writeItems having troubles writting booleans!

flacohenao, 2020-05-28:
Hi Ghuilherme!

Writting a boolean value on a bit separated area like this:

`itemGroup.writeItems("DB2,X0.1", true)`

throws me this error:

```
ERROR WRITE TAG NodeS7Error: Write error [0x7]: Data type inconsistent
    at S7ItemGroup.writeItems (C:\Users\ACER\Documents\NODE PROJECTS\tot-bend\node_modules\nodes7\src\s7itemGroup.js:548:27)   
    at processTicksAndRejections (internal/process/task_queues.js:97:5) {
  code: 7,
  info: undefined
```

I also have tried passing the value as a (1 or 0), even string as ('true', 'false')

I guess that the error there didnt catch any info coz its obvious that you know what's the item you attempting to write

gfcittolin, 2020-05-29:
Hi, I can confirm the issue. It is related to writing to boolean items only. This have been fixed in the past, but got intentionally reverted for some reason. I'm checking why that was the case, and will provide unit tests that cover this to prevent from happening in the future.

flacohenao, 2020-05-29:
Thank you very much!!

flacohenao, 2020-06-02:
@gfcittolin Hi!!!

is the commit the final fix? can you please explain what was causing the issue?

commit 9b8d7d67 (2020-06-01): Fixes #114

Fix again bug wrongly reverted by 3193ee04 and add test case for it