plcpeople/nodeS7 #125: boolen write problem ???

bayramozgn, 2022-01-13:
I can't write data to bool value. Do you have an example?

WirelessSwitcher, 2022-06-23:
Same here, I'm trying to read from "TEST1: M20.0" and "TEST2: M20.1", but the software returns "BAD: 255".
If you had any luck, let me know.

plcpeople, 2022-06-24:
I added an example that involves writing a boolean and added mention of the debug option in the example in the README.

If you can post your code, and some console logs with debug set to true, or even some logs without it set to true as errors will be reported anyway, we can likely get you going.

WirelessSwitcher, 2022-06-24:
I'm sorry, I wasn't able to find the example of writing into a Boolean in the readme, but I found the overall example 

Here's the code:
// Initializing required libraries:
const express = require("express")
const path = require("path");
const fs = require("fs");
var bodyParser = require("body-parser");

// Initializing App:
const app = express();
const port = process.env.PORT || 3000;

// Setting path for public director
const static_path = path.join(__dirname, "public");
app.use(express.static(static_path));
app.use(express.urlencoded({ extended: true }));
app.use(bodyParser.json());

// Server Setup
app.listen(port, () => {
	console.clear();
	console.log(`Server is running at ${port}\n`);
});

/* *** NodeS7 *** */
const nodes7 =  require("nodes7");                                       // Adds the driver to the applicaiton
const conn = new nodes7;
var doneReading = false;
var doneWriting = false;

// Declare variable table
var variables = {
    TEST1: "M20.0",        // Bit at M20.0
    //TEST2: "M20.1",        // Bit at M20.1
}

// Start comms
conn.initiateConnection({ port: 102, host: '172.16.1.101', rack: 0, slot: 1 }, connected); // slot 2 for 300/400, slot 1 for 1200/1500
// conn.initiateConnection({port: 102, host: '192.168.0.2', localTSAP: 0x0100, remoteTSAP: 0x0200, timeout: 8000, doNotOptimize: true}, connected);
// local and remote TSAP can also be directly specified instead. The timeout option specifies the TCP timeout.

function connected(err) {
    if (typeof(err) !== "undefined") {
        // We have an error. Maybe the PLC is not reachable.
        console.log(err);
        process.exit();
    }
    conn.setTranslationCB(function(tag) { return variables[tag]; }); // This sets the "translation" to allow us to work with object names
    conn.addItems(['TEST1']);//, 'TEST2']);
    //conn.addItems('TEST6');
    //conn.removeItems(['TEST2', 'TEST3']); // We could do this.
    //conn.writeItems(['TEST5', 'TEST6'], [ 867.5309, 9 ], valuesWritten); // You can write an array of items as well.
    //conn.writeItems('TEST1', [true], valuesWritten); // You can write a single array item too.
    conn.readAllItems(valuesReady);
}

function valuesReady(anythingBad, values) {
    console.log(values);
    if (anythingBad) { console.log("SOMETHING WENT WRONG READING VALUES!!!!"); }
    console.log(values);
    doneReading = true;
    if (doneWriting) { process.exit(); }
}

function valuesWritten(anythingBad) {
    if (anythingBad) { console.log("SOMETHING WENT WRONG WRITING VALUES!!!!"); }
    console.log("Done writing.");
    doneWriting = true;
    if (doneReading) { process.exit(); }
}

/* *** SENDING TO PLC *** */

// Receives the message from the client
app.post("/", function(req){
	// Prints the content of the message from the client
	console.log(req.body.cmd);

    if(req.body.cmd == "write"){
        console.log("Writing in M20.0");
        //conn.addItems([TEST1])
        conn.writeItems(["TEST1"], true, valuesWritten);
    }

    if(req.body.cmd == "clear"){
        console.clear();
    }

});

Here's the terminal:
Server is running at 3000

[29065,460418100 172.16.1.101 S1] TCP Connection Established to 172.16.1.101 on port 102
[29065,460779300 172.16.1.101 S1] Will attempt ISO-on-TCP connection
[29065,461271900 172.16.1.101 S1] Using rack [0] and slot [1]
[29065,469672200 172.16.1.101 S1] ISO-on-TCP Connection Confirm Packet Received
Server is running at 3000

[29510,382455300 172.16.1.101 S1] TCP Connection Established to 172.16.1.101 on port 102
[29510,382659300 172.16.1.101 S1] Will attempt ISO-on-TCP connection
[29510,383270200 172.16.1.101 S1] Using rack [0] and slot [1]
[29510,388076900 172.16.1.101 S1] ISO-on-TCP Connection Confirm Packet Received
[29510,434933200 172.16.1.101 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
[29510,435324900] Translation OK
[29510,436018900 172.16.1.101 S1] Adding TEST1
[29510,444359700] PDU type (byte 8) was returned as 2 where the response PDU of 3 was expected.
[29510,444638900] Maybe you are requesting more than 240 bytes of data in a packet?
[29510,444973900] <Buffer 03 00 00 13 02 f0 80 32 02 00 00 00 02 00 00 00 00 81 04>
[29510,445659800 172.16.1.101 S1] ConnectionReset has been called to set the reset as pending
[29511,953115500 172.16.1.101 S1] READ TIMEOUT on sequence number 2
[29511,954217600 172.16.1.101 S1] Processing an undefined packet, likely due to timeout error
[29511,954655600 172.16.1.101 S1] Timeout error - zero length packet
[29511,955051100 172.16.1.101 S1] Received a ZERO RESPONSE Processing Read Packet due to unrecoverable packet error
[29511,955771200 172.16.1.101 S1] Calling reset from readResponse as there is one pending
[29511,956491100 172.16.1.101 S1] ResetNOW is happening
[29511,956693300 172.16.1.101 S1] Connection cleanup is happening
[29511,957669700 172.16.1.101 S1] Attempting to connect to host...
{ TEST1: 'BAD 255' }
SOMETHING WENT WRONG READING VALUES!!!!
{ TEST1: 'BAD 255' }
[29511,965277600 172.16.1.101 S1] TCP Connection Established to 172.16.1.101 on port 102
[29511,965629400 172.16.1.101 S1] Will attempt ISO-on-TCP connection
[29511,965861600 172.16.1.101 S1] Using rack [0] and slot [1]
[29511,975182600 172.16.1.101 S1] ISO-on-TCP Connection Confirm Packet Received
[29512,30353400 172.16.1.101 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.``

plcpeople, 2022-06-24:
This looks very similar to what is seen on an S7-1200 or S7-1500 if you haven't enabled GET/PUT in TIA Portal and downloaded the hardware configuration to the PLC following the change.  (Which unfortunately requires putting the PLC in STOP).  Are you able to do this?

WirelessSwitcher, 2022-06-24:
Yes, sure I can do it, I'll test it again Monday and let you know

WirelessSwitcher, 2022-06-28:
I am trying to configure TIA Portal, but it's being trickier than I assumed (I am used with Modbus, where every global has a Modbus mask readily available).

I see TIA requires the "other device" to be declared, so I used the Desktop-PC w/ 1 port, gave it my pc's IP.
Now I have to configure the PUT and GET blocks/DBs, but I'm finding it hard... I tried Siemens forums and google, but all I could get were PLC2PLC examples. I also found some videos of a nice guy called "chunz"in youtube, but he never shows the TIA  configuration, only the other side.

Does anyone out there know a place where I can find a tutorial or nice explanation in order to make it work?

plcpeople, 2022-06-28:
You don't need to use the PUT and GET blocks, those are only needed for PLC to PLC communication on the "active" partner of the connection.  The "passive" side, the PLC, doesn't need any code, but it does need you to open the "Device Configuration" (Hardware Configuration) in TIA Portal and go to "Protection and Security" and "Connection Mechanisms".  Under there you have to check the box that says "Permit Access with PUT/GET" and download the hardware again.  That's it - no code.  This is necessary whether you need nodeS7 or many other non-Siemens HMI packages to talk to the PLC, or even for another PLC to access it using the PUT and GET blocks.

plcpeople, 2022-06-28:
[image]

WirelessSwitcher, 2022-06-28:
Wow, excellent! Thank you very much. It worked like a gem! Now it's about developing the front end!

Once again, thanks for sharing this excellent project!

plcpeople, 2022-07-01:
Good to hear it's working.  Good luck with the front end for your project.

closed 2022-07-01 by plcpeople