plcpeople/nodeS7 #129: TCP ECCONRESET

anacarolina-ms, 2022-10-06:
Hey! I'm working with your node on node-red for reading from a plc S7-1200, rack 0 and slot 1 with external functions. But every time I send a readAllItems I get the response I want but the connection resets every time and never closes, even with dropConnection and connectionCleanup... I've been trying to solve it for a few days now, if anyone has any ideas...

```javascript
var devices = msg.payload; // devices example: {"delta":0,"name":"10.255.3.214","rack":0,"slot":1,"data":"Q0.0","ip":"10.255.3.214","port":102}

msg = [];

let nodes7 = global.get('nodes7');
var conn = new nodes7({ globalTimeout: 2000 });
var doneReading = false;

devices.forEach((/** @type {{ modbus_id: any; modbus_ip: any; modbus_port: any; address: any; quantity: any; }} */ dev) => {
   var variables = {
      DATA: dev.data
   };
   
   conn.initiateConnection({ port: dev.port, host: dev.ip, rack: dev.rack, slot: dev.slot, debug: false }, connected);
   function connected(err) {
      if (typeof (err) !== "undefined") {
         // PLC FORA DO ALCANCE
         //console.log(err);
         node.error({
            "ip": dev.ip,
            "port": dev.port,
            "message": "ERRO! Falha na leitura do host " + dev.ip + "!",
            "error": err.code
         });
         flow.set("aguarda", false);
         conn.dropConnection();
         conn.connectionCleanup();
      }
      conn.setTranslationCB(function (tag) { return variables[tag]; });
      conn.addItems('DATA');
      conn.readAllItems(valuesReady);
   }

   function valuesReady(anythingBad, values) {
      if (anythingBad) { console.log("ERRO! Falha na leitura do host."); flow.set("aguarda", false); conn.dropConnection(); conn.connectionCleanup(); }
      //console.log(values);
      if (values.DATA != "BAD 255") {
         node.send({
            "topic": "TCP Response",
            "host": dev.ip,
            "port": dev.port,
            "rack": dev.rack,
            "slot": dev.slot,
            "data": dev.data,
            "payload": values.DATA
         });
      }
      doneReading = true;
      conn.dropConnection();
      conn.connectionCleanup();
   }
   
   
});

return msg;

```

anacarolina-ms, 2022-10-06:
Debug output:
```
0|node-red  | [6357,358840334] Initiate Called - Connecting to PLC with address and parameters:
0|node-red  | [6357,360228467] { port: 102, host: '10.255.3.214', rack: 0, slot: 1, debug: false }
0|node-red  | [6357,361356973] Connection cleanup is happening
0|node-red  | [6357,362302985 10.255.3.214 S1] Attempting to connect to host...
0|node-red  | [6357,369721605 10.255.3.214 S1] TCP Connection Established to 10.255.3.214 on port 102
0|node-red  | [6357,370300805 10.255.3.214 S1] Will attempt ISO-on-TCP connection
0|node-red  | [6357,370687512 10.255.3.214 S1] Using rack [0] and slot [1]
0|node-red  | [6357,389771024 10.255.3.214 S1] ISO-on-TCP Connection Confirm Packet Received
0|node-red  | [6357,441426946 10.255.3.214 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
0|node-red  | [6357,441951617] Translation OK
0|node-red  | [6357,442560191 10.255.3.214 S1] Adding DATA
0|node-red  | [6357,478182400] We Caught a read/write error ECONNRESET - will DISCONNECT and attempt to reconnect.
0|node-red  | [6357,478722852] ConnectionReset is happening
0|node-red  | [6357,479273668] ConnectionReset is happening
0|node-red  | [6358,984437130] ResetNOW is happening
0|node-red  | [6358,985126067] Clearing an earlier scheduled reset
```

anacarolina-ms, 2022-10-06:
Obs.: I want to read just one time and one address and close connection

minhbonghot, 2022-10-07:
Hi carolina
You can use process.exit()
Here for example:

`var nodes7 = require('nodes7'); // This is the package name, if the repository is cloned you may need to require 'nodeS7' with uppercase S
var conn = new nodes7;

var variables = {
    machineNo: "DB1,C0.10",
};

conn.initiateConnection({ port: 102, host: '192.168.0.1', rack: 0, slot: 1, debug: false }, connected);

function connected(err) {
  if (typeof(err) !== "undefined") {
    // We have an error. Maybe the PLC is not reachable.
    console.log(err);
    process.exit();
  }
  conn.setTranslationCB(function(tag) { return variables[tag]; });
  conn.addItems("machineNo");
  console.log("-----------------Init done-------------")
  conn.readAllItems(valuesReady)
}

function valuesReady(anythingBad, values) {
  if (anythingBad) { console.log("SOMETHING WENT WRONG READING VALUES!!!!"); }
  console.log(values);
  process.exit()
}`

anacarolina-ms, 2022-10-07:
Hi minhbonghot, thanks for your reply!

When I use the external module 'process.exit()' inside node-red, apparently the process it ends is the execution of node-red itself, so this solution doesn't work for me...

If I could just disable reconnection maybe the problem would be solved, I don't know.

plcpeople, 2022-10-07:
Hi Ana, can you please change this to turn more debugging output on and then post the output again:
` var conn = new nodes7({ globalTimeout: 2000 });
to
` var conn = new nodes7({ globalTimeout: 2000, debug: true });

I don't have node red set up right now so I edited your code to look like this in index.js and I ran it standalone, and it disconnects from the PLC and reconnects properly:
```
let nodes7 = require('nodes7');
var conn = new nodes7({ globalTimeout: 2000, debug: true });
var doneReading = false;
var net = require('net');
var server = net.createServer();    

server.listen(9000, function() {    
  console.log('server listening to %j', server.address());  
});

conn.initiateConnection({ port: 102, host: "10.8.1.50", rack: 0, slot: 1, debug: true }, connected);
function connected(err) {
  if (typeof (err) !== "undefined") {
	 // PLC FORA DO ALCANCE
	 console.log(err);
	 conn.dropConnection();
	 conn.connectionCleanup();
  }
  conn.setTranslationCB(function (tag) { return tag; });
  conn.addItems('Q0.0');
  conn.readAllItems(valuesReady);
}

function valuesReady(anythingBad, values) {
  if (anythingBad) { console.log("ERRO! Falha na leitura do host."); conn.dropConnection(); conn.connectionCleanup(); }
  //console.log(values);
  if (values != "BAD 255") {
console.log(values);
  }
  doneReading = true;
  // If I comment out the next 2 lines the connection stays active
  conn.dropConnection();
  conn.connectionCleanup();
  setTimeout(() => {
	conn.initiateConnection({ port: 102, host: "192.168.1.2", rack: 0, slot: 1, debug: true }, connected);
  }, 3000);
}
```

plcpeople, 2022-10-07:
Hi Ana, another thing I was going to mention:

There are a couple of projects that implement nodes7 within node-red already:
https://github.com/st-one-io/node-red-contrib-s7
https://github.com/Hilscher/node-red-contrib-s7comm

Would these help?

anacarolina-ms, 2022-10-10:
Hey! So, I managed to make it work for a while, I tested it for 3 days in a row to do consecutive readings from a PLC, however after a while the reading stopped working and I started receiving ECCONRESET again, this time without reading...

I believe that the problem is not the PLC because I monitored direct readings by MODBUS/TCP and they worked during these 3 days...

New debug output:
```
0|node-red  | [254366,355884] Initiate Called - Connecting to PLC with address and parameters:
0|node-red  | [254366,881148] { port: 102, host: '10.255.3.214', rack: 0, slot: 1, debug: false }
0|node-red  | [254366,1491359] Connection cleanup is happening
0|node-red  | [254366,1847507 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [254366,2180426 10.255.3.214 S1] Clearing write PacketTimeouts
0|node-red  | [254366,3311369 10.255.3.214 S1] <initiating a new connection Mon Oct 10 2022 13:20:11 GMT-0300 (Horário Padrão de Brasília)>
0|node-red  | [254366,3774341 10.255.3.214 S1] Attempting to connect to host...
0|node-red  | [254366,9753906 10.255.3.214 S1] TCP Connection Established to 10.255.3.214 on port 102
0|node-red  | [254366,10404795 10.255.3.214 S1] Will attempt ISO-on-TCP connection
0|node-red  | [254366,10890579 10.255.3.214 S1] Using rack [0] and slot [1]
0|node-red  | [254366,17295614 10.255.3.214 S1] ISO-on-TCP Connection Confirm Packet Received
0|node-red  | [254366,24938626] We Caught a read/write error ECONNRESET - will DISCONNECT and attempt to reconnect.
0|node-red  | [254366,25622953] Connection cleanup is happening
0|node-red  | [254366,25921809 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [254366,26214675 10.255.3.214 S1] Clearing write PacketTimeouts
0|node-red  | [254366,47073125] Initiate Called - Connecting to PLC with address and parameters:
0|node-red  | [254366,47506565] { port: 102, host: '10.255.3.214', rack: 0, slot: 1, debug: false }
0|node-red  | [254366,48094849] Connection cleanup is happening
0|node-red  | [254366,48362455 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [254366,48563029 10.255.3.214 S1] Clearing write PacketTimeouts
0|node-red  | [254366,49127980 10.255.3.214 S1] <initiating a new connection Mon Oct 10 2022 13:20:11 GMT-0300 (Horário Padrão de Brasília)>
0|node-red  | [254366,49468034 10.255.3.214 S1] Attempting to connect to host...
0|node-red  | [254366,54414729 10.255.3.214 S1] TCP Connection Established to 10.255.3.214 on port 102
0|node-red  | [254366,54838064 10.255.3.214 S1] Will attempt ISO-on-TCP connection
0|node-red  | [254366,55085670 10.255.3.214 S1] Using rack [0] and slot [1]
0|node-red  | [254366,63911032 10.255.3.214 S1] ISO-on-TCP Connection Confirm Packet Received
0|node-red  | [254366,71997067] We Caught a read/write error ECONNRESET - will DISCONNECT and attempt to reconnect.
0|node-red  | [254366,73648534] Connection cleanup is happening
0|node-red  | [254366,74032495 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [254366,74346767 10.255.3.214 S1] Clearing write PacketTimeouts

```
The reading process only works again when I restart my Raspiberry Pi.

I started the readings at: 2022-10-07T15:15:06.9640000-03:00
And the last one was: 2022-10-08T09:49:59.7200000-03:00

And about the nodes for Node-RED, I tested them but they do not allow me to dynamically change the Host IP and Port as I do in this code.

Is there any connection limit? Because it seems to me that it works for 2 days then it stops working as if it has reached a limit.

Anyway, thanks for the help, I'll keep researching and trying to fix this. I'm still open to suggestions!

anacarolina-ms, 2022-10-10:
Output before restarting Raspberry Pi:
```

0|node-red  | [323,685113244] Initiate Called - Connecting to PLC with address and parameters:
0|node-red  | [323,685749182] { port: 102, host: '10.255.3.214', rack: 0, slot: 1, debug: true }
0|node-red  | [323,687229600 10.255.3.214 S1] Connection cleanup is happening
0|node-red  | [323,687755069 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [323,688827153 10.255.3.214 S1] Clearing write PacketTimeouts
0|node-red  | [323,690795071 10.255.3.214 S1] <initiating a new connection Mon Oct 10 2022 13:46:33 GMT-0300 (Horário Padrão de Brasília)>
0|node-red  | [323,691423249 10.255.3.214 S1] Attempting to connect to host...
0|node-red  | [323,703902842 10.255.3.214 S1] TCP Connection Established to 10.255.3.214 on port 102
0|node-red  | [323,704503728 10.255.3.214 S1] Will attempt ISO-on-TCP connection
0|node-red  | [323,705047427 10.255.3.214 S1] Using rack [0] and slot [1]
0|node-red  | [323,712378162 10.255.3.214 S1] ISO-on-TCP Connection Confirm Packet Received
0|node-red  | [323,757600803 10.255.3.214 S1] Received PDU Response - Proceeding with PDU 240 and 3 max parallel connections.
0|node-red  | [323,759229346] Translation OK
0|node-red  | [323,759816638 10.255.3.214 S1] Reading All Items (readAllItems was called)
0|node-red  | [323,760632211 10.255.3.214 S1] Adding or Removing { arg: 'DATA', action: 'add' }
0|node-red  | [323,762026900 10.255.3.214 S1] Adding DATA
0|node-red  | [323,763486485 10.255.3.214 S1] self.globalReadBlockList 0 parts is 1 offset is 0 MBR is 220
0|node-red  | [323,764226641 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [323,766004143 10.255.3.214 S1] Calling SRP from RAI
0|node-red  | [323,766579091 10.255.3.214 S1] SendReadPacket called
0|node-red  | [323,766999717 10.255.3.214 S1] seqNum is 2
0|node-red  | [323,767541488 10.255.3.214 S1] Sending Read Packet With Sequence Number 2
0|node-red  | [323,776068734 10.255.3.214 S1] Received 5 bytes of S7-data from PLC.  Sequence number is 2
0|node-red  | [323,776613527 10.255.3.214 S1] ReadResponse called
0|node-red  | [323,777293162 10.255.3.214 S1] Time is 0 seconds and 9.8 ms.
0|node-red  | [323,777958163 10.255.3.214 S1] Address Q0.0 has value false and quality OK
0|node-red  | [323,778690143 10.255.3.214 S1] We are calling back our readDoneCallback.
0|node-red  | [323,781397228 10.255.3.214 S1] Connection cleanup is happening
0|node-red  | [323,782558792 10.255.3.214 S1] Clearing read PacketTimeouts
0|node-red  | [323,783143115 10.255.3.214 S1] Clearing write PacketTimeouts

```
Now it's working again, but I can't say for how long.

liulysk, 2022-10-11:
Hello, I'm dealing with similar issues and what I found out is that if you keep connection alive (won't drop it and )
it will add new Used connection to S7 communication pool every time you make a function call 
[image]

When you reach the end of the pool then you get ECCONRESET and can only recover from it by restarting NR.
I wonder if it is possible to keep connection alive and have some connections pool defined by argument?

```
function valuesReady(anythingBad, values) {
  if (anythingBad) { 
      //console.log("SOMETHING WENT WRONG READING VALUES!!!!"); 

      node.warn("SOMETHING WENT WRONG READING VALUES!!!!");
      conn.dropConnection(); 
      conn.connectionCleanup();
  }
  console.log(values);
  msg.payload = values;
  node.send(msg);

  conn.dropConnection();
  conn.connectionCleanup();
```

plcpeople, 2022-10-13:
@anacarolina-ms can you please check open TCP connections between the Pi and PLC using netstat on the Raspberry Pi or by looking at the active connection information in TIA Portal or the PLC web page, if enabled?  I set up a similar flow (I pasted the flow posted by @liulysk in issue #130 and it has run for a few hours talking to an S7-1200 and so far seems to release all connections.

anacarolina-ms, 2022-10-14:
Hi @plcpeople thanks for your help! Sending the outputs of netstat from my Raspberry Pi above.

netstat:
```
Conexões Internet Ativas (sem os servidores)
Proto Recv-Q Send-Q Endereço Local          Endereço Remoto         Estado
tcp        0      0 localhost:1413          localhost:4512          ESTABELECIDA
tcp        0      0 10.255.1.15:ssh         10.255.2.129:58845      ESTABELECIDA
tcp        0      0 10.255.1.15:ssh         10.255.2.129:58805      ESTABELECIDA
tcp        0      0 localhost:4512          localhost:1413          ESTABELECIDA
tcp        0      0 10.255.1.15:1880        10.255.2.129:59100      ESTABELECIDA
Domain sockets UNIX ativos (sem os servidores)
Proto RefCnt Flags       Type       State         I-Node   Caminho
unix  2      [ ]         DGRAM                    12109    /run/user/1000/systemd/notify
unix  4      [ ]         DGRAM      CONECTADO     7983     /run/systemd/notify
unix  2      [ ]         DGRAM                    7997     /run/systemd/journal/syslog
unix  21     [ ]         DGRAM      CONECTADO     8004     /run/systemd/journal/dev-log
unix  7      [ ]         DGRAM      CONECTADO     8006     /run/systemd/journal/socket
unix  2      [ ]         DGRAM      CONECTADO     11994    /tmp/.vncserver-license/0.493
unix  2      [ ]         DGRAM      CONECTADO     15908    /tmp/dhcpcd-nexxus/libdhcpcd-wpa-1031.0
unix  2      [ ]         DGRAM      CONECTADO     15909    /tmp/dhcpcd-nexxus/libdhcpcd-wpa-1031.1
unix  4      [ ]         DGRAM      CONECTADO     11895    /var/run/wpa_supplicant/wlan0
unix  2      [ ]         DGRAM                    11931    /var/run/wpa_supplicant/p2p-dev-wlan0
unix  3      [ ]         STREAM     CONECTADO     14768483
unix  3      [ ]         STREAM     CONECTADO     13811
unix  2      [ ]         DGRAM                    11856
unix  3      [ ]         STREAM     CONECTADO     14767455
unix  3      [ ]         DGRAM      CONECTADO     12189
unix  3      [ ]         STREAM     CONECTADO     15026
unix  3      [ ]         STREAM     CONECTADO     14082
unix  3      [ ]         STREAM     CONECTADO     14768486
unix  3      [ ]         STREAM     CONECTADO     16398
unix  3      [ ]         STREAM     CONECTADO     10925
unix  3      [ ]         STREAM     CONECTADO     8548     /run/systemd/journal/stdout
unix  2      [ ]         STREAM     CONECTADO     14375244
unix  3      [ ]         STREAM     CONECTADO     15713
unix  3      [ ]         STREAM     CONECTADO     11212
unix  3      [ ]         STREAM     CONECTADO     17701    /run/user/1000/pulse/native
unix  3      [ ]         STREAM     CONECTADO     14768467
unix  3      [ ]         STREAM     CONECTADO     14442
unix  3      [ ]         STREAM     CONECTADO     14768463
unix  3      [ ]         STREAM     CONECTADO     14375328
unix  3      [ ]         STREAM     CONECTADO     16676    /run/user/1000/bus
unix  3      [ ]         STREAM     CONECTADO     14766436
unix  3      [ ]         STREAM     CONECTADO     17645    /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONECTADO     15066    @/tmp/.X11-unix/X0
unix  3      [ ]         STREAM     CONECTADO     15023
unix  3      [ ]         STREAM     CONECTADO     14767441
unix  3      [ ]         STREAM     CONECTADO     13029
unix  3      [ ]         STREAM     CONECTADO     13809
unix  3      [ ]         STREAM     CONECTADO     11346    /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONECTADO     13175    /run/dbus/system_bus_socket
unix  3      [ ]         STREAM     CONECTADO     14766438
unix  3      [ ]         STREAM     CONECTADO     15702
unix  3      [ ]         STREAM     CONECTADO     14636    /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONECTADO     13791
unix  3      [ ]         STREAM     CONECTADO     14767442
unix  3      [ ]         STREAM     CONECTADO     14768432
unix  3      [ ]         STREAM     CONECTADO     14768491
unix  3      [ ]         STREAM     CONECTADO     13813
unix  3      [ ]         DGRAM      CONECTADO     14376096
unix  3      [ ]         STREAM     CONECTADO     16678
unix  3      [ ]         STREAM     CONECTADO     11735    /run/dbus/system_bus_socket
unix  3      [ ]         STREAM     CONECTADO     17634    /run/dbus/system_bus_socket
unix  3      [ ]         STREAM     CONECTADO     14599
unix  3      [ ]         STREAM     CONECTADO     16521
unix  3      [ ]         STREAM     CONECTADO     17558    @/tmp/.X11-unix/X0
unix  3      [ ]         STREAM     CONECTADO     10941
unix  3      [ ]         STREAM     CONECTADO     11732
unix  3      [ ]         STREAM     CONECTADO     15849
[... 326 lines cut ...]
    TCPWantZeroWindowAdv: 1
    TCPSynRetrans: 31
    TCPOrigDataSent: 12324631
    TCPHystartTrainDetect: 9
    TCPHystartTrainCwnd: 440
    TCPHystartDelayDetect: 5
    TCPHystartDelayCwnd: 887
    TCPACKSkippedChallenge: 1
    TCPKeepAlive: 257
    TCPDelivered: 12781126
    TCPAckCompressed: 2737
    TcpTimeoutRehash: 295
    TcpDuplicateDataRehash: 16
    TCPDSACKRecvSegs: 785
IpExt:
    InMcastPkts: 131230
    OutMcastPkts: 400
    InBcastPkts: 1241219
    InOctets: 4532163791
    OutOctets: 1879727761
    InMcastOctets: 12393958
    OutMcastOctets: 43918
    InBcastOctets: 108892480
    InNoECTPkts: 16722651
```